technyon/nuki_hub #4: _mqttMaxBufferSize filling most of the heap

rodriguezst, 2022-06-14:
I am implementing TLS support to MQTT to encrypt communications. It was working fine with version 2.6 (before opener support) but does not work now when rebasing to master.
It seems that there is no heap space available for the SSL due to an increase of the MQTT buffer.
`_mqttMaxBufferSize = 16384;`
If I decrease the value of _mqttMaxBufferSize from 16k to the default value (256bytes) it works fine.
Have you measured the maximum needed buffer? Are those 16k really needed?

technyon, 2022-06-14:
16k is a bit excessive, I had to increase it due to the CSV-data not getting publishes if too many devices are present. I think it could be reduced to something more reasonable. Feel free to try; the question though is how many devices to support for presence detection.

I'll look into it later.

technyon, 2022-06-14:
I've looked into it and there was a lot of memory to be saved. I never had the need to look into it, so I just kept it like this. I think you should have even more space available now than before 2.6.

rodriguezst, 2022-06-15:
Thanks. It works fine now. I have submitted a PR with the TLS encryption changes.

closed 2022-06-15 by rodriguezst