technyon/nuki_hub #9: Nuki Bridge API

rueckix, 2022-06-19:
Anybody tried/thought of implementing the Nuki Bridge Api (https://developer.nuki.io/page/nuki-bridge-http-api-1-13/4) ?
That would alllow for a drop-in replacement for the official bridge, while preserving all features of home automation systems. E.g., openhab (https://www.openhab.org/addons/bindings/nuki/).

rodriguezst, 2022-06-19:
@rueckix this other repo implements what you want:  https://github.com/dauden1184/RaspiNukiBridge

rueckix, 2022-06-20:
Thanks... except that I thought it'd be handy to have the bridge API on the ESP32, of course ;-)

technyon, 2022-06-20:
Having a quick look at the Bridge API, I'd say it's possible but would require a substantial amount of work. Not only writing the code, but also testing with existing smart home solutions and so on. Probably not going to happen too soon.

rueckix, 2022-06-20:
I agree, not trivial, but doable if we have enough flash left. Have to yet order and set up my dev kit.

technyon, 2022-06-20:
I'd say there's enough left

################## Size Summary ##################
Program Size: 1545758 of 1966080 bytes (78%)
Data Size: 50260 of 327680 bytes (15%)
##################################################

A big part of what's used already is the BLE lib I think.

technyon, 2022-11-29:
Closing due to inactivity, it's not on my to-do list. Let me know if this is still relevant.

closed 2022-11-29 by technyon