technyon/nuki_hub #11: Wifi Reconnect Issue

abubakaraz, 2022-06-20:
Hi, 

As already reported in the ioBroker forum, I have problems with the wifi connection. Version 4.2 has not fixed the problem. Can I provide you with any logs ?

Thanks
abu

technyon, 2022-06-20:
Does your ESP module have a USB connector? If yes, logs via serial would be helpful. Connect with a terminal program (e. g. for Windows hterm) to the ESP32 (set speed to 115200).

FYI, I'll probably won't be available until end of week to check for replies.

technyon, 2022-06-25:
Please try Version 4.4

rastos, 2022-06-26:
Hi @technyon , I'm facing probably same issue, I can replicate it when I restart my router, the ESP loses connection and never reestablishes it again,
I connected to console and this is what I get on loop, other devices have no issues connecting back to wifi after router reboot, also if I powercycle the hub connects back withtout issue.

I'm on latest 4.4 version

```
Reconnect failed
Network not connected. Trying reconnect.
*wm:[2] [EVENT] WIFI_REASON:  201
*wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND
*wm:[2] [EVENT] WIFI_REASON:  201
*wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND
Reconnect failed
```
cheers

technyon, 2022-06-26:
Hi,

It's really hard to me to figure out what's going to. I leave the reconnect to WifiManager, so once a disconnect event is received it should try to reconnect. I've tried several thing:

- I have routers. Disable all then switch one on again
- Disable the active router, so the ESP has to switch to the other one
- Change the channel, again the ESP has to reconnect

Is there anything special about your Wifi setup?

Other than that you can use the network timeout instead, which will restart the ESP when the timeout is reached.

technyon, 2022-06-27:
I've read some reports of Ubiquity being problematic with the ESP32 Wifi power save mode. Unfortunately when I disable power saving for Wifi, the ESP crashes.

Is setting the network timeout to something short a workaround for you (Might be worth trying something like 10 seconds)?

abuaz01, 2022-06-28:
Unfortunately the workaround does not work.
I still lose the connection after about 1-2 hours

technyon, 2022-06-28:
Can you test the attached binary? Basically as soon as the MQTT connection drops, it reboots the ESP, no timeout.

The network timeout option checks the status of the wifi connection, not MQTT.

[nuki_hub.zip](https://github.com/technyon/nuki_hub/files/9003564/nuki_hub.zip)

apfelflo89, 2022-06-28:
I just flashed it, I'll let you know tomorrow if it's still reachable and connected. Thank you

technyon, 2022-06-29:
Please collect  some logs. As a last resort I could add a reboot-timer, but I'd prefer to fix the issue properly. I really don't understand why the ESP doesn't reboot, an MQTT disconnect is easy to detect.

abuaz01, 2022-06-29:
Hi, Mac overwrite the logs after some time :/
Any Idea to collect and save? 

This is the last error log: 
abuaz01$ screen /dev/cu.usbserial-0001 115200
*wm:[2] NUM CLIENTS:  0
*wm:[2] NUM CLIENTS:  0
*wm:[2] NUM CLIENTS:  0
*wm:[2] NUM CLIENTS:  0
*wm:[2] NUM CLIENTS:  0
*wm:[2] NUM CLIENTS:  0
*wm:[2] NUM CLIENTS:  0

technyon, 2022-06-29:
Interesting ... I think this isn't an error, but your ESP has restarted into the WifiManager config portal. Did you use the last binary I posted?

abuaz01, 2022-06-29:
No, the logs are from the binary from yesterday. 
i already flash the new binary that you post today. The esp is still reachable after 3 hours. I will update you tomorrow.

abuaz01, 2022-06-30:
Good Morning, 
looks good. My ESP is now since 10 hours reachable.

technyon, 2022-06-30:
OK, let's wait just a little bit if it stays that way. If so, I'll properly implement this and make it configurable.

technyon, 2022-06-30:
Version 4.6 is released. You have to enable "Restart on disconnect" to have the same behavior as 4.5-testing. Please report back so I can close the issue.

apfelflo89, 2022-07-01:
I'm on 4.6 and my Nuki-hub is still connected for one day

technyon, 2022-07-01:
Did you enable "Restart on disconnect" ?

H0useman, 2022-07-01:
with 4.6 is my ESP has always accessible via WiFi
Restart on disconnect is enable

good job :)

abuaz01, 2022-07-03:
My ESP is again not reachable after 2 days. After the reboot everything works again... very frustrating :(

technyon, 2022-07-03:
Since I'm running out of ideas, I've added an option to configure a timer to restart the ESP after a certain amount of time. No Idea why the Wifi on the ESP is so shaky.

H0useman, 2022-07-03:
Maybe a ping interval from esp to the gateway will help

mundschenk-at, 2022-07-03:
> Since I'm running out of ideas, I've added an option to configure a timer to restart the ESP after a certain amount of time. No Idea why the Wifi on the ESP is so shaky.

ESPhome reboots after a (configurable) timeout when the WiFi connection stalls (by default: 15 minutes). Not sure how they detect it.

technyon, 2022-07-04:
There's already a network timeout option. This is driven though by the MQTT client disconnecting, but that happens quite fast after WiFi disconnects. Unfortunately it doesn't help much for most users having trouble ... and strange thing is it works for many without problems at all.

There's one other possibility: The ESP is somehow crashed or stuck and doesn't process anything anymore, but only logs would tell. Not too likely though, there's a task watchdog active already, so if a task is stuck it should reboot.

It's quite annoying to fix this as I can't reproduce the behavior on my end, so I can only guess what's going wrong.

rastos, 2022-07-04:
Hi
this issue is weird, not sure if everybody is facing it in the same way, but I'm attaching log, maybe it will help
I did run ESPhome on the same ESP32 that I'm running this bridge on and never experienced issues with sensor running on it on my home assistant.  so I think? this should exclude HW. I tried replacing power source just to be sure.

the log shows start, when the device was stuck in the config portal loop, I couldn't find the SSID of the config portal, so I assume somehow the access to wifi module is lost?
next i restarted the device with button, line 41 you can see it get's IP address (redacted), then it fails, restarts, manages to connect to wifi, then fails again and wifi never comes back until I did another HW reset.

could there be an overflow that causes issues with wifi connection even after software reboot? I'm not familiar with coding for ESP so I'm not able to help :[

how ever as you can see this does not happen 100% of time, sometimes the device stays up for 24h sometimes it crashes after 2h, the most stable I got it is when I don't access the web just restart it and let it be. 

now with 4.8 I have the device on 120minutes restart period, and if that does the trick, i'll be probably happy for home use.

[nuki.log](https://github.com/technyon/nuki_hub/files/9039600/nuki.log)

technyon, 2022-07-04:
This is very strange. Is WiFi reception good near your ESP?

Also what's concerning is that there was an unhandled exception so the ESP restarted after connecting. 

I'm using an ESP32 Mini, this model to be exact:
https://www.amazon.de/-/en/gp/product/B093GQGJCV/

technyon, 2022-07-05:
Update on the issue: I can at least reproduce it now. Earlier versions seemed to work fine.

... so I did a bit of digging thorugh the commit history. I found a few change I made to the task scheduling which can have all kind of side effects. Long story short, I have a hunch this could the cause of the problems (which I think isn't one isolated problems, but a few which lead to the same result: ESP is not reachable).

Please test the attached binary, if possible both with and without the "Restart on disconnect" option. I've reverted the task scheduling changes in that version.

[nuki_hub.zip](https://github.com/technyon/nuki_hub/files/9049043/nuki_hub.zip)

apfelflo89, 2022-07-06:
I switched to ethernet with a W5500 now, I installed nuki-hub in the basement and cannot really log there. I really appreciate every troubleshooting you did so far, but LAN is much more reliable than the wifi handling of the ESP

technyon, 2022-07-07:
All four ESPs are still up and running. I think this has fixed the issue ... I will prepare a release later.

todde99, 2022-07-07:
I can confirm that nuki_hub-4.9-testing is running for 10 hours without any issue.

technyon, 2022-07-07:
Version 5 is out. No need to flash if you're using 4.9-testing except for a new version number.

technyon, 2022-07-07:
P.S.: I have disabled both restart on disconnect and restart timer for testing. Might be a good idea to disable those, since they are workarounds and the problem should be fixed now.

abuaz01, 2022-07-07:
Hi @technyon 
Thank you for your incredible work and support!
My ESP has been running for 24 hours now without any problems. 

I would close the issue Saturday if that is ok for you. 
Thanks

technyon, 2022-07-08:
Does anyone still need the "Restart timer" option after the fixes? Otherwise I'd remove it for the next release.

rastos, 2022-07-08:
@technyon I can report that on 5.0 my esp is stable for 24+h without restart timer, or any other timeout options,

thank you for your excellent work!

closed 2022-07-09 by abuaz01