technyon/nuki_hub #22: OTA update attempt makes web server return 401

mundschenk-at, 2022-07-17:
When I try to update via the OTA mechanism, I get the page indicating that it will automatically reload when the update is ready. A few seconds later, the browser tries to reload, but the authentication credentials are not accepted (so I'm getting the popup to enter different ones).

This happens sometimes randomly, but with OTA updates, it is reproducible for me. Here's two simulated requests using telnet:

Normal:
```shell
$ telnet <hostname> 80
Trying <IP>...
Connected to <hostname>.
Escape character is '^]'.
GET / HTTP/1.1
HOST: nukihub.local
Authorization: Basic <base64-encoded-credentials>
HTTP/1.1 200 OK
Content-Type: text/html
Content-Length: 1034
Connection: close

<HTML><HEAD><meta name='viewport' content='width=device-width, initial-scale=1'><link rel='stylesheet' href='/inter.css'><link rel='stylesheet' href='/new.css'><TITLE>NUKI Hub</TITLE></HEAD><BODY><br><h3>Info</h3>
<table><tr><td>Hostname</td><td>nukihub</td></tr><tr><td>MQTT Connected</td><td>Yes</td></tr><tr><td>NUKI Lock paired</td><td>Yes</td></tr><tr><td>NUKI Lock state</td><td>locked</td></tr><tr><td>Firmware</td><td>5.2</td></tr></table><br><br><h3>MQTT and Network Configuration</h3><form method="get" action="/mqttconfig"><button type="submit">Edit</button></form><BR><BR><h3>NUKI Configuration</h3><form method="get" action="/nukicfg"><button type="submit">Edit</button></form><BR><BR><h3>Credentials</h3><form method="get" action="/cred"><button type="submit">Edit</button></form><BR><BR><h3>Firmware update</h3><form method="get" action="/ota"><button type="submit">Open</button></form><br><br><h3>WiFi</h3><form method="get" action="/wifi"><button type="submit">Restart and configure wifi</button></form></BODY></HTML>Connection closed by foreign host.
```

After update attempt:
```shell
$ telnet <hostname> 80
Trying <IP>...
Connected to <hostname>.
Escape character is '^]'.
GET / HTTP/1.1
HOST: nukihub.local
Authorization: Basic <base64-encoded-credentials>
HTTP/1.1 401 Unauthorized
Content-Type: text/html
WWW-Authenticate: Basic realm="Login Required"
Content-Length: 0
Connection: close
```

mundschenk-at, 2022-07-17:
So I now tried again with the USB console connected to my computer. After a false start (didn't actually select a file), the update seemed to go through without the credentials thing, but the reported firmware ist still 5.2. The output from the console was pretty sparse (the three `Invalid file for OTA upload` are from one attempt, by the way):

```shell
Invalid file for OTA upload
Invalid file for OTA upload
Invalid file for OTA upload
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
Unable to setup HASS. Invalid config received.
Attempting MQTT connection
MQTT: Connecting with user: mosquitto_nuki
MQTT connected
```

technyon, 2022-07-17:
Hi,

regarding the Invalid file error message, the check is this:

```
    if(upload.filename == "")
    {
        Serial.println("Invalid file for OTA upload");
        return;
    }
```

This check is to prevent starting an update when no file has been selected.

mundschenk-at, 2022-07-17:
> Hi,
> 
> regarding the Invalid file error message, the check is this:
> 
> ```
>     if(upload.filename == "")
>     {
>         Serial.println("Invalid file for OTA upload");
>         return;
>     }
> ```
> 
> This check is to prevent starting an update when no file has been selected.

Yeah, what I found interesting that a single invalid upload resulted in three lines on the console.

technyon, 2022-07-17:
It's a handler method that is called by the web server. Hard to say exactly why there are three callbacks, my guess would be your browser is trying to send the not selected file multiple times.

I'll see if I can reproduce the behavior by interrupting an update in the middle.

technyon, 2022-07-17:
Can you check with Version 5.4? There was a nasty buffer overflow bug in the authentication code which can lead to undefined behavior which leads to bugs like yours.

mundschenk-at, 2022-07-17:
Thanks, I will check it out. A minor thing (should be harmless), but the upper bound for the credUser is 20 chars in the settings, so the stop-gap zero-terminator should be at index 21, no? The current fix uses 31 for both user and PW.

technyon, 2022-07-17:
Yes the size of the user should be changed to 21 ... or the character limit increased to 30. Shouldn't matter though, it's 10 bytes wasted, which isn't nice but shouldn't lead to any bug.

mundschenk-at, 2022-07-17:
So I updated to 5.4 and then tried to trigger the error using the OTA update mechanism. 

Good news: I did not experience issues with the credentials.
Bad news: I still was not able to complete an OTA upgrade. Things silently failed with the 5.4 binary, so I assume there is a version check somewhere. I then tried with the 5.3 binary (i.e. a downgrade) and got a bit farther, but things stalled (the browser reports a terminated connection).

Here are the log files (note that the 5.4 upgrade attempts were completely silent, i.e not even a `handleFileUpload` message):

```text
MQTT connected
Nuki start pairing
Nuki paired
Attempting MQTT connection
MQTT: Connecting with user: <mqtt-user>
MQTT connected
Unable to setup HASS. Invalid config received.
handleFileUpload Name: /nuki_hub.bin
handleFileUpload Name: /nuki_hub.bin
1436
BeginOTA
2872
4308
5744
7180
8616
10052
11488
handleFileUpload Name: /nuki_hub.bin
```

technyon, 2022-07-18:
Somehow the upload died. Does this always happen?

Ilovegym66, 2022-07-18:
Hi, I'd the same issues on my 4 ESP's. 3 of them need 5-7 times than the OTA run's through without n issue. One of them I can't update from 5.2. to 5.3/5.4, the same error appears.. I tested some different browsers, reboot / powercycle the esp.. etc.. same effect again.. 
The OTA from 5.1 to 5.2 run's without an issue.

mundschenk-at, 2022-07-18:
> Somehow the upload died. Does this always happen?

For me, pretty much. Though I've not tried that often with 5.4 (maybe 5-6 times all in all?).

So without looking at the code, does the upload not happen before the actual flashing process starts? I would have assumed that the file is first uploaded completely and only when the flashing starts we see the. `BeginOTA` message?

technyon, 2022-07-18:
The ESP has two partitions for the firmware, each can contain a firmware. Once you see BeginOTA, it starts to receive data and writes it to the unused partition. When the upload completes (EndOTA), the ESP switches the active firmware to the partition the file has just been uploaded to, end then restarts.

So flashing does start start as soon as data is received, but the switching between partitions prevents firmware from interrupted uploads to be used. It's pretty safe like this, at least you don't delete your current firmware.

mundschenk-at, 2022-07-18:
Thank you for the explanation, @technyon. I've got several ESPHome devices running and OTA flashing them rarely needs more than two tries (and even that mostly when I do it from my tablet and accidentally change orientation, resulting in a reload of the HA app).

technyon, 2022-07-18:
Strange. I've tried OTA several times now and it worked every time at the first try.

@[Ilovegym66](https://github.com/Ilovegym66) Anything that has changed in your configuration since it worked last time. Browser, Network? Code-wise OTA is pretty much the same.

Ilovegym66, 2022-07-19:
Network not changed, only Browser (update of chrome)
Can it be, that the flash cycle is limited at the ESP some time?

technyon, 2022-07-19:
Well there's one minute time to flash, that should be more than enough. Currently it's just a timer, which is not optimal, unless the upload is extremely slow. Can you observe in the develop console of the browser what's going on (Press F12 then switch to Network)?

Ilovegym66, 2022-07-19:
Found it ! it's a timeout problem with the network timeout and restart timeout config in the mqtt-config-tab :-) 
I disabled both with -1 and de-select the checkmark for the reboot, now the ota works like expected.
My config was before 120, checkmark on, 60 see pic below, please.
[image]

technyon, 2022-07-19:
Thanks I'll look into it. Probably just disable network timeout when OTA starts.

I see you have the restart timer option set, is it still necessary?

Ilovegym66, 2022-07-19:
1) yes, would be nice to get no timeout errors any more.
2) I don't know, never tested another option.. :-) In the post the first versions had some network issues.. since this option exist, I'd enabled it and the esp works very good. ( all 4 )

technyon, 2022-07-19:
Please test with the attached firmware. Network timeout is disable as soon as OTA starts.

Although unrelated, could test disabling the "Restart timer". I'd like to remove that option, it was only a last resort fix.

[nuki_hub.zip](https://github.com/technyon/nuki_hub/files/9144168/nuki_hub.zip)

mundschenk-at, 2022-07-19:
I just got some M5Stack Atom Lite and flashed 5.3 using `esptool` to the test the OTA feature. I was able to flash 5.4 on the first try, very different from my experience with the D1 Mini clone I used previously. I've now configured it the same as the previous one and put it in its "hidden" location (i.e. not optimal regarding wifi coverage) – what can I say, `5.4-tst-1` flashed without a hitch as well. 

The only configuration difference was that I had now disabled `Restart on disconnect` (I had never enabled either the restart time nor the network timeout). However, disabling that did not seem to help when I tried with the old one previously. Maybe it is faulty in some way.

Ilovegym66, 2022-07-20:
5.4-tst-1 works fine, after the ota update the esp lost the wifi config, the lock was paired.

closed 2022-07-24 by technyon