technyon/nuki_hub #23: Using "Lock'n'go" does not change the lock state in Home Assistant

mundschenk-at, 2022-07-19:
It appears that when using the "Lock'n'go" feature (e.g. via a double click of the lock button), the state of the `lock` entity in Home Assistant is not updated (i.e. neither the `lock` nor `unlock` action register in the logbook, it looks as if the lock remained closed).

technyon, 2022-07-20:
@rodriguezst Care to look into this? I have no Home Assistant setup to check this.

mundschenk-at, 2022-07-20:
Anything I can help with? I don't know too much about MQTT, but I can look if pointed in the right direction. I am assuming that "Lock'n'go" sends a different BLE message that not mapped to unlocked/locked states?

mundschenk-at, 2022-07-20:
I don't think it's the speed of messages - the lock'n'go message appears to not do be caught or processed at all. The locking only happens 20 seconds later, surely that can't be too fast to process the unlocking message first?

technyon, 2022-07-20:
Maybe also include "unlatching" and map it to unlocked. I haven't seen it very often, but just for the sake of completeness maybe include it.

mundschenk-at, 2022-07-20:
> Unlatch isn't caught all the time because it changes back and forth too quick, but I think lockngo should be slow enough. You could just try and observe how the state changes.

If you mean the MQTT state? I did, it didn't change at all. That's why I concluded that it isn't only the HA mapping after all.

technyon, 2022-07-24:
I mean the reported state in MQTT. When you execute an unlatch action, the reported state goes to "unlatching" to "unlatched" and back to "unlocked", but often the firmware and thus MQTT is unaware of unlatching and unlatched because it happens too fast.

mundschenk-at, 2022-07-24:
> I mean the reported state in MQTT. When you execute an unlatch action, the reported state goes to "unlatching" to "unlatched" and back to "unlocked", but often the firmware and thus MQTT is unaware of unlatching and unlatched because it happens too fast.

Ah, OK, I would not really care about the "unlatched" state as knowing when the door is unlocked would be enough (I have a Z-Wave door sensor to see when it was opened). 

Here's the current sequence of events:

1. The door is locked.
2. I push the button twice.
3. The door unlocks itself.
4. I open & close the door (or not, for testing).
5. After 20 seconds or so, the Nuki locks itself again.

During this procedure, there is currently no visible state change, the lock remains "locked" the whole time according to MQTT.

technyon, 2022-08-06:
Hi,

I finally had some time to setup  my own HA testing instance. I've fixed the issue a bit differently than suggestet: I created a new "binaryState" that maps the other states to either locked or unlocked, and changed the HA config accordingly.

Could you test if this fixes the issue (you'll eventually have to disable and enable the HA integration --> clear and enter HA path).

[nuki_hub.zip](https://github.com/technyon/nuki_hub/files/9275358/nuki_hub.zip)

mundschenk-at, 2022-08-06:
I'll check it out, sure!

mundschenk-at, 2022-08-06:
> Could you test if this fixes the issue (you'll eventually have to disable and enable the HA integration --> clear and enter HA path).

Works great! Both normal unlocking/locking and lock'n'go update the state of the lock.

technyon, 2022-08-06:
Fixed in release 5.6

closed 2022-08-06 by technyon

mundschenk-at, 2022-09-09:
I can't do a controlled experiment right now, but is it possible that recent changes have resulted in a regression regarding the recognition of Lock'n'go states?