technyon/nuki_hub #24: Changing any network setting resets HA device/entity customization

mundschenk-at, 2022-07-25:
Changing anything in the `/mqttconfig` screen results in all device/entity customizations (friendly name, icon, entity slug) being reset because the HA discovery data is removed even when the HA topic was not changed at all. This is because all config keys are always present when saving (also ensuring a restart on every save because `configChanged` is always true).

technyon, 2022-07-26:
I have added the suggested fix (not committed yet). Can you check if it works like this?

[nuki_hub.zip](https://github.com/technyon/nuki_hub/files/9192374/nuki_hub.zip)

mundschenk-at, 2022-07-26:
I updated the FW and clicked "Save" without actually changing any setting - the HA entities stayed safely in their customized form.

technyon, 2022-07-26:
ok I'll prepare a release then

closed 2022-07-26 by technyon