technyon/nuki_hub #25: Protokoll vom Lock als JSON senden

tt-tom17, 2022-07-28:
Hallo,

besteht die Möglichkeit nach der Statusänderung vom Lock die Daten authorizationId, authorizationName, state und trigger als json über MQTT zu senden?

Gruß Tom

technyon, 2022-07-28:
Das dürfte machbar sein ... was hast Du vor?

tt-tom17, 2022-07-28:
Ich habe festgestellt das die einzelnen Topic nicht synchron aktualisiert werden lt. Zeitstempel in ioBroker. Ich möchte eine Protokoll JSON bauen mit den letzten 14 Aktivitäten. Zur Zeit trigger ich auf den State und lass ein Script die Daten lesen und als JSON ausgeben.

technyon, 2022-07-29:
Das ganze wird nich synchron gesendet weil es mehrere requests brauch, den letzten Log-Eintrag  abzufragen, das ist an sich komplizierter als nötig. Der Name wird zuletzt geschickt, Du könntest alles abfragen sobald da ein Update kommt.

Ansonsten hat's mich jetzt leider mit Covid erwischt, von daher werd ich die nächsten Tage erstmal nichts machen.

tt-tom17, 2022-07-29:
Alles klar, dann trigger ich mal auf den Namen und beobachte mal das Ergebnis. Dir wünsche ich gute Besserung und einen leichten Verlauf.

tt-tom17, 2022-07-30:
Hi, ich habe noch ein paar Tests gemacht und ein Script geschrieben um die Daten mit zu loggen. Der Trigger auf den Namen klappt nicht, da er nur aktualisiert wird, wenn er sich ändert. Der einzige Topic den man triggern kann ist der State, da dieser sich immer ändert. Ich hänge mal mein log mit ran, da kannst du die Zeitstempel sehen zu den einzelnen Topics.
Diese ändern sich nur, wenn auch der Wert / Inhalt sich änder.  Der Name "Nuki Lock" habe ich im Script erstellt, da unter seiner ID nur ein Leerstring gesendet wird.

Gute Besserung 

`
script.js.common.Nuki.Protokoll: id: xxxx967295 - 11:15:56, Name: Nuki Lock - 11:15:56, Zustand: locked - 11:20:46, Auslöser: manual - 11:15:56
script.js.common.Nuki.Protokoll: id: xxxx967295 - 11:15:56, Name: Nuki Lock - 11:15:56, Zustand: unlocked - 11:21:01, Auslöser: button - 11:21:01
script.js.common.Nuki.Protokoll: id: xxx1645 - 11:21:57, Name: Tom - 11:21:57, Zustand: locked - 11:21:56, Auslöser: system - 11:21:56
script.js.common.Nuki.Protokoll: id: xxx1645 - 11:21:57, Name: Tom - 11:21:57, Zustand: unlocked - 11:22:27, Auslöser: system - 11:21:56
script.js.common.Nuki.Protokoll: id: xxx1647 - 11:23:27, Name: NukiHub - 11:23:27, Zustand: locked - 11:23:26, Auslöser: system - 11:21:56
script.js.common.Nuki.Protokoll: id: xxx1647 - 11:23:27, Name: NukiHub - 11:23:27, Zustand: unlocked - 11:24:19, Auslöser: system - 11:21:56
`

technyon, 2022-07-30:
Das geht :). Versuch das mal wie folgt:

on({ id: "mqtt.0.nuki.lock.authorizationName", change: 'any' }, () => 
{
  // Dein Code here
});

tt-tom17, 2022-07-30:
okay, habe Script umgestellt und werde das Protokoll beobachten.

tt-tom17, 2022-07-31:
so läuft es, habe noch ein weiteren Fehler bei mir gefunden. Mein MQTT - Adapter war falsch eingestellt.

closed 2022-07-31 by tt-tom17