technyon/nuki_hub #26: Support ESPHome

github-k8n, 2022-07-28:
Would it be possible to implement this as a custom module in esphome?

i.e. providing custom switch and/or custom sensor components
https://esphome.io/components/switch/custom.html
https://esphome.io/components/sensor/custom.html

https://github.com/thegroove/esphome-custom-component-examples
https://esphome.io/components/external_components.html

technyon, 2022-07-28:
While probably not impossible to do, it's not in the scope of this project. I have little experience so far with esphome, and I'm not sure how esphome manages bluetooth since the hardware is a shared ressource (other esphome modules might want to access the bluetooth chip as well).

github-k8n, 2022-07-28:
maybe this one would help?
https://esphome.io/components/ble_client.html

Not sure how much of your code is wifi/webserver/bluetooth  connection handling (which you would not need in this case) and what part is regarding the communication with the lock (which could maybe use the above module and just do the communication? (not entirely clear on that..))

mundschenk-at, 2022-07-28:
While I don't think it is currently anywhere near as polished as this firmware, someone has written an ESPHome component for the Nuki lock: https://github.com/uriyacovy/ESPHome_nuki_lock

zanna-37, 2022-07-28:
This would be beyond amazing 😍
Please consider having a look at it if you can.

zanna-37, 2022-08-10:
https://github.com/esphome/esphome/pull/3398 this new feature might also help. =)

technyon, 2022-09-17:
Hi, I'm closing this writing an esp home module is not on my to-do list. If a developer needs help integrating the NUKI-side of this, feel free to ask.

closed 2022-09-17 by technyon

ruimarinho, 2023-01-02:
I second this request! @technyon basically you get a webserver, mqtt support, homeassistant auto-discovery, OTA support, Wifi manager, GPIO/Ethernet, and more out of the box. It would be a lot less code to maintain because only Nuki-specific logic would have to be added.

zanna-37, 2023-01-03:
On top of that, the ESP can do more. You can attach other sensors, etc. you are not limited to one thing only.

bcutter, 2023-01-15:
> I second this request! @technyon basically you get a webserver, mqtt support, homeassistant auto-discovery, OTA support, Wifi manager, GPIO/Ethernet, and more out of the box. It would be a lot less code to maintain because only Nuki-specific logic would have to be added.

Oh yes. Summarizing all current open-source attempts of integrating Nukis into HA, an ESPHome based would really be the nearly perfect solution. I would also love to see Nuki Hub relying on ESPHome instead of MQTT.