technyon/nuki_hub #31: Compilation information

jmsucasas, 2022-08-21:
This job is incredible,  I have lot of time searching a project like this!.
Is it possible to add some information to compile an create the bin file using the expressiff tools?

Thank you in advance,
BR

technyon, 2022-08-21:
Check my other project which uses the same toolchain:

https://github.com/technyon/coffeetimer

You should configure the partition table to "minspiffs"

rodriguezst, 2022-08-21:
You can also have a look at the GitHub Actions workflow file with all the steps needed to prepare the environment and compile the project at .github/workflows/build.yml

jmsucasas, 2022-08-23:
It appears to be solved git-clonning nuki-ble library.  The directory nuki_hub seems to be empty when I git-clone the full project.

technyon, 2022-08-23:
The nuki ble library is included as a submodule. You have to checkout with --recurse-submodules. It's not a good idea to manually clone the library into the nuki_ble directory.

closed 2022-08-27 by technyon