technyon/nuki_hub #34: OTA Update

Dragon0501, 2022-09-09:
Please add the possibility to use directly the offered 7z file. The update doesn't work with the extracted nuki_hub.bin

technyon, 2022-09-09:
The ESPs flash is too small to unpack a 7z file on the fly. I'm also not sure how that's supposed to fix OTA ... can you elaborate on what's actually happening? Sometimes it doesn't work on the first try, but you can just give it another try.

Dragon0501, 2022-09-09:
What should I do... Tried it three times, but nothing changes. Still on 5.9

technyon, 2022-09-09:
How good is the Wifi reception where the ESP is located? Can you try moving it closer to the router for the update?

technyon, 2022-09-15:
OP not responding, closing issue.

Dragon0501, 2022-09-15:
You should not close it. The problem still exist. I don't what I can do. Trying to update doesn't change the firmware version...

technyon, 2022-09-16:
Hi. I didn't hear back from you. If you want it fixed, please work with me, can you answer my question above?

technyon, 2022-09-16:
ok ... it's a bit hard to tell why it's not working. Can you connect via usb cable to the ESP and get the logs? You can use for example HTERM to get the logs, baud rate is 115200.

Dragon0501, 2022-09-17:
I can do that. What do I have to do? Just connecting and everything works by itself or what do I have to do?

technyon, 2022-09-17:
- Connect the ESP to a PC or Laptop
- Open HTERM
- Select the correct COM Port
- Select 115200 Baud
- Connect

Now you should see logs from the ESP. if the output isn't formatted corrected, try changing the "Newline at" setting.

Dragon0501, 2022-09-18:
Problem solved, but I don't know how... Connected it to hterm and update succeeded.
Sad but the log file is attached...

[Log.txt](https://github.com/technyon/nuki_hub/files/9593526/Log.txt)

technyon, 2022-09-18:
Well, observer effect I guess. I can only guess from here ... you could check the power supply and/or the USB cable you're (if it's different from the one you use for HTERM).

Dragon0501, 2022-09-18:
Usb cable was the same. Maybe I'll try again with next version