technyon/nuki_hub #38: Port nuki_hub to Olimex ESP32-POE-ISO Board

pilo1337, 2022-09-25:
Hi,

i was wondering porting this app to an Olimex ESP32 PoE Board?

this has several benefits:
- LAN Mode with PoE, no need to wire power
- All in one Board
- No wifi connection drop issues

Unfortunately, the board uses a LAN8710A Chip in RMII Mode for managing Ethernet, which is not supported by the current firmware.

the board is well documentated, maybe we can write a port for this particular board?

technyon, 2022-09-25:
Hi, it should be possible to port this to the Olimex board. However, there are quite a few more boards with LAN connectivity and this is a project I'm running and financing privately - I unfortunately can't buy every hardware out there.

pilo1337, 2022-09-25:
yeah, i understand that, no prob.

Maybe i can help you with stuff? i looked into your code, looks like the network stack is "modular", maybe we can write another stack for LAN8710A ?

technyon, 2022-09-25:
Sure ... thinking about maybe it's enough to just wire a LAN8710A module the same way it's wired on the board? They are fairly cheap.

pilo1337, 2022-09-26:
should be. i tried flashing the firmware in hope to use the esp32 wifi, but unfortunately, the gpio pin for eth activation (gpio26) is wired to the nic chip. this leads to deactivating the wifi. which IDE do you used for developing this code?

technyon, 2022-09-26:
I'm using clion, but if your question is how to compile the code it's a cmake project. Have a look at my other project "coffeetimer", there's a howto for compiling.

Not sure about the GPIO ... I had to pick one so it was 26. We could make it configurable in software, but that has the risk of someone changing to the setting to LAN and then have an ESP that isn't reachable because the hardware is missing.

Maybe have two settings for LAN:
- Let the user choose which GPIO switches on LAN
- Let the user select a driver

This way also issue #36 could be addressed.

pilo1337, 2022-09-26:
i have no pratical experience with cmake programs. i managed to install arduino sdk, cmake and toolchain, but i got error when trying to compile: 
philipp@PF1RKYGV:~/nuki_hub-master$ Picked up JAVA_TOOL_OPTIONS:
cmake -D CMAKE_TOOLCHAIN_FILE=/home/philipp/Arduino-CMake-Toolchain-master/Arduino-toolchain.cmake .
CMake Error at /home/philipp/Arduino-CMake-Toolchain-master/Arduino/System/BoardBuildTargets.cmake:472 (message):
  Arduino library BLE could not be found in
  /home/philipp/nuki_hub-master;/home/philipp/nuki_hub-master;/home/philipp/Arduino;/home/philipp/.arduino15/packages/esp32/hardware/esp32/1.0.6;/home/philipp/arduino-1.8.19
Call Stack (most recent call first):
  /home/philipp/Arduino-CMake-Toolchain-master/Arduino/System/BoardBuildTargets.cmake:713 (find_arduino_library)
  /home/philipp/Arduino-CMake-Toolchain-master/Arduino/System/BoardBuildTargets.cmake:680 (_add_internal_arduino_library)
  /home/philipp/Arduino-CMake-Toolchain-master/Arduino/System/BoardBuildTargets.cmake:183 (_link_ard_lib_list)
  CMakeLists.txt:93 (target_link_arduino_libraries)

CMake Error at /home/philipp/Arduino-CMake-Toolchain-master/Arduino/System/BoardBuildTargets.cmake:166 (message):
  DNSServer is not a CMake target
Call Stack (most recent call first):
  CMakeLists.txt:111 (target_link_arduino_libraries)

CMake Error at CMakeLists.txt:111 (target_link_arduino_libraries):
  cmake_policy PUSH without matching POP

-- Configuring incomplete, errors occurred!
See also "/home/philipp/nuki_hub-master/CMakeFiles/CMakeOutput.log".
See also "/home/philipp/nuki_hub-master/CMakeFiles/CMakeError.log".

any short hint?

technyon, 2022-09-26:
- you have to checkout with submodules. Run: git submodule update --init --recursive
- You should create a subdirectory for building, usually "build":

mkdir build
cd build
cmake -D CMAKE_TOOLCHAIN_FILE=/home/philipp/Arduino-CMake-Toolchain-master/Arduino-toolchain.cmake ..

Also, please use my fork of the arduino toolchain:

https://github.com/technyon/Arduino-CMake-Toolchain

Which version of the Arduino core for ESP32 did you install?

pilo1337, 2022-09-26:
yeah, the toolchain file worked for me. i used esp32 version 1.0.6, because all of the 2.0 vers does not find a suitable CXX compiler:
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:5 (project):
  The CMAKE_CXX_COMPILER:

    /home/philipp/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++

  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Configuring incomplete, errors occurred!
See also "/home/philipp/nuki_hub/build/CMakeFiles/CMakeOutput.log".
See also "/home/philipp/nuki_hub/build/CMakeFiles/CMakeError.log".

pilo1337, 2022-09-26:
i dont get it, now it compiled, but i dont know how to flash it, the directive "upload-nuki_hub" does not work, because no bootloader is generated.

another hint?

technyon, 2022-09-26:
Strange with the compiler, I didn't have that problem. Maybe just set a symbolic link? But for your purposes, 1.0.6 is fine. Pre 2.0 the WiFi problems were much more pronounced, but apart from that it's the same.

You could just copy the bootloader from the official release (basically all .bin files).

I'm using ninja though instead of make, maybe that would fix your problems. Are you on Debian or Ubuntu? In that case run:

apt install ninja-build

Then remove all files from the cmake build directory and add the option "-GNinja" to the cmake command line.

pilo1337, 2022-09-26:
yeah, i was dumb. got it to work.
the initialisation of the lan module seems quite simple, should an easy task.
am i right that i have to add another "adapter" in networkDevices/ and put the init stuff in network.cpp?

or am i missing something?

technyon, 2022-09-26:
Yes, create a new device in networkDevices, then instantiate it in network.cpp.

The question is how to switch between W5500 and LAN8720. What do you think about selecting driver and GPIO in the web portal?

pilo1337, 2022-09-27:
yeah, i would use wifi as standard, because every esp32 does have it. after first boot, you can decide in the webinterface which driver you want to use.
this gpio selection is nice, but in this scenario you should use this gpio as a "rescue", so setting the gpio to ground will force to default (wifi), in case you have misconfigured anything.

technyon, 2022-09-27:
It's a bit hard to tell like this what's the issue. Could you fork my project and check in your code so I can have a look at it? Also which library are you using to communicate with the LAN8170?

I don't think it's related to cmake ... if you have problems with cmake it's usually about getting your code to compile.

ruimarinho, 2023-01-31:
Getting LAN8170 support would be amazing so I could replace my WiFi-based ESP32 by one of these Olimex boards with PoE built-in! Any additional progress so far @pilo1337?

technyon, 2023-02-03:
Hi,

I've added support for the M5Stack Atom POE which uses the W5500 chip:

https://docs.m5stack.com/en/atom/atom_poe

If I find some time I'll try to add their other ESP with Ethernet:

https://shop.m5stack.com/products/esp32-ethernet-unit-with-poe

The latter is considerably cheaper, but lacks a USB port so you have to connect a USB-serial converter.

technyon, 2023-02-04:
@pilo1337 Can you share the full code ?

technyon, 2023-02-22:
The Olimex is finally supported !

closed 2023-02-22 by technyon

ruimarinho, 2023-02-23:
Amazing @technyon 👏 ! Will the PoE port be automatically detected or will it still require a WiFi connection to enable the ethernet component?

technyon, 2023-02-23:
You have to first connect it to wifi and then select olimex hardware. I don't really have a good way to detect that it's running on an olimex board.

ruimarinho, 2023-02-24:
A different image with this pre-defined values would be great. If you use platform.io, you'd just need to compile the image for a different environment with specific build tags. Example: https://github.com/OpenEVSE/ESP32_WiFi_V4.x/blob/master/platformio.ini

enkama, 2024-01-09:
> You have to first connect it to wifi and then select olimex hardware. I don't really have a good way to detect that it's running on an olimex board.

Hey. I know pretty old issue. But is the way still to "select olimex hardware" because I cant find where I would select it so it runs over lan/poe instead of wlan?!