technyon/nuki_hub #39: Cannot connect to the smartlock

0rsa, 2022-09-28:
Hello,
I cannot pair the smartlock

> *wm:[2] Connection result: WL_CONNECTED
> *wm:[1] AutoConnect: SUCCESS 
> *wm:[2] Connected in 4634 ms
> *wm:[1] STA IP Address: xxx.xxx.xxx.xxx
> WiFi connected: xxx.xxx.xxx.xxx
> Host name: nukihub
> MQTT Broker: xxx.xxx.xxx.xxx
> NUKI Lock enabled
> Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: yes
> NUKI Opener disabled
> Presence detection timeout (ms): 60000
> Attempting MQTT connection
> MQTT: Connecting without credentials
> MQTT connected
> Nuki start pairing
> Nuki start pairing
> Nuki start pairing
> Nuki start pairing
> Nuki start pairing

Nuki and ESP32 are separated by 5 centimeters.
I use the latest Nuki 2.0 firmware 2.12.4

I tried to factory reset the smartlock but still trying to pair without success.

Do you have an idea?

Thank you

0rsa, 2022-09-28:
I tried to register as an app but the issue persists.

technyon, 2022-09-28:
Did you pair your smartphone already?

0rsa, 2022-09-29:
First yes the smartphone was paired to the phone.
It was not after the factory reset

technyon, 2022-09-29:
ok, just asking to make sure pairing in general is working.

It's hard to tell, usually it just works. How close is the ESP to the lock?

technyon, 2022-09-29:
I'm somewhat out of ideas ... I'd have to add some log messages to see what's really going on.

0rsa, 2022-09-29:
Ok, thank you.
From my side, I ordered a different model of my current ESPs to try if it can the responsible of this issue.

technyon, 2022-09-29:
Please try with the attached firmware and send logs. When flashed the version string is "6.0-dbg-connect".

[nuki_hub-dbg-connect.zip](https://github.com/technyon/nuki_hub/files/9678084/nuki_hub-dbg-connect.zip)

0rsa, 2022-09-30:
Oh, new logs:

[289304][D][NukiBle.cpp:102] pairNuki(): No nuki in pairing mode found
[289305][D][NukiBle.cpp:107] pairNuki(): pairing result 0
[294285][D][NukiBle.cpp:229] onResult(): Found nuki in pairing state:  addr: xx:xx:xx:xx:xx:xx
Nuki start pairing
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
[302732][W][NukiBle.cpp:138] connectBle(): BLE Connect failed, retrying
[332746][W][NukiBle.cpp:138] connectBle(): BLE Connect failed, retrying
[362759][W][NukiBle.cpp:138] connectBle(): BLE Connect failed, retrying

technyon, 2022-09-30:
The error packet count mismatch comes from the bluetooth stack, but is normal, it happens sometimes.

Your ESP somehow has problems though with bluetooth communication. In the first log, it didn't receive the iBeacon from the lock, in the second it tried to connect but failed, so the pairing process wasn't initiated.

Why this is happening is hard to tell. I'd try another ESP as you suggested. No point in meddling with bluetooth parameters, the first log is about receiving a beacon, and there's no way to improve reception from software side, you'd need to change the antenna ... that's not the problem here.

StefanTR96, 2022-10-10:
Had the same Problem ... I paired it with the App, selected Configure Bridge, powered on the ESP32 and pushed the button for 10 seconds (The light started after about 5 seconds)  ... The App told "success" and the Web Interface of the ESP32 told "Paired"

yaronkof, 2022-10-15:
@technyon here are my logs
[esp-web-tools-logs.txt](https://github.com/technyon/nuki_hub/files/9792239/esp-web-tools-logs.txt)

I set my lock to paring mode, and the esp just got stuck in a Nuki start pairing loop

technyon, 2022-10-15:
It's very strange ... I just unpaired my lock and paring worked immediately. I've compiled the latest firmware with enhanced logging, can you give it a try? When successfully flashed, the versions reads "6.0-dbg-1".

[nuki_hub-dbg.zip](https://github.com/technyon/nuki_hub/files/9792310/nuki_hub-dbg.zip)

yaronkof, 2022-10-15:
> It's very strange ... I just unpaired my lock and paring worked immediately. I've compiled the latest firmware with enhanced logging, can you give it a try? When successfully flashed, the versions [...]

how do I flash this ver?

gonzalezcalleja, 2022-10-15:
Same issue here, can't pair.

- Previously I have been using RaspiNukiBridge (https://github.com/dauden1184/RaspiNukiBridge) in a RPI for a long time.
- Nuki Smart Lock firmware 3.3.5 on Nuki Smart Lock 3.0 device.
- Tested with 2 ESP32 devices with same result.
- Tested with nukihub 6.0, 6.0-dbg-1 and 5.15.
- Tested with "register as app" on and off.
- Restored to factory settings the Nuki smart lock.
- Tested with [nuki_hub-dbg-connect.zip](https://github.com/technyon/nuki_hub/files/9678084/nuki_hub-dbg-connect.zip) from this issue @technyon:

```
Attempting MQTT connection
MQTT: Connecting without credentials
MQTT connect failed, rc=-2
Free Heap: 112864
Nuki start pairing
[963732][D][NukiBle.cpp:102] pairNuki(): No nuki in pairing mode found
[963733][D][NukiBle.cpp:107] pairNuki(): pairing result 0
```

(in loop)

technyon, 2022-10-16:
Same as with the logs before, for some reason the ESP isn't picking up the beacon from the lock, so it doesn't initiate the pairing procedure. I have no explanation really, here it just works. I'm on firmware 3.3.5 on a NUKI 3.0 too.

Can you try to enable the presence detection and see if it's reporting anything? If it doesn't wobk try my other firmware:

https://github.com/technyon/blescanner

This just reports beacons from devices in the area. Just to make sure your ESP receives any beacons at all and that bluetooth is working.

technyon, 2022-10-16:
> how do I flash this ver?

- download a release and unzip it
- replace the nuki_hub.bin in the unpacked release with the debug bin
- follow the instructions in the readme

gonzalezcalleja, 2022-10-17:
> Same as with the logs before, for some reason the ESP isn't picking up the beacon from the lock, so it doesn't initiate the pairing procedure. I have no explanation really, here it just works. I'm on firmware 3.3.5 on a NUKI 3.0 too.
> 
> Can you try to enable the presence detection and see if it's reporting anything? If it doesn't wobk try my other firmware:
> 
> https://github.com/technyon/blescanner
> 
> This just reports beacons from devices in the area. Just to make sure your ESP receives any beacons at all and that bluetooth is working.
> 
> P.S.: Just saw that it worked after a lot of tries. Something's up with the bluetooth communication. Using the app on the smartphone works without problems?

Hi,

Yes, the lock works over bluetooth, I use every day from 2 smartphone and from Raspinuk (https://github.com/dauden1184/RaspiNukiBridge).

I have tested blescanner and the ESP32 doesn't show the Nuki Lock device, but I can see other devices, is strange because with hcitool lescan from a RPI I always receive the beacons from the Nuki.

Thanks!

gonzalezcalleja, 2022-10-17:
With this script https://gist.github.com/elliotlarson/1e637da6613dbe3e777c (ibeacon-scan.bash) I can see a lot of beacons every second from a Tesla card ... could be this too much beacons for an ESP32 an is ignoring the Nuki beacons?

gonzalezcalleja, 2022-10-18:
Tested today with other ESP32 boards and worked an paired at the first try:

Nuki start pairing
Nuki paired
Nuki lock state: locked
Reading config. Result: 1
Reading advanced config. Result: 1

The new board is from az-Delivery and the older one with bluetooth problems from DUBEUYEW:

[image]

Thanks for your support @technyon, finally was a hardware problem with the ESP32 boards.

technyon, 2022-10-20:
Hi,

thanks for feedback. Maybe it's time to compile a list of recommended hardware ... still I have no explanation really and without being able to reproduce the issue it's really hard to fix. At least I can give the advice to try a different board now.

SergioWarriorPeru, 2022-10-21:
Hi, a question...where is the option "register as an app"?

technyon, 2022-10-21:
In the configuration portal go to NUKI Configuration, in the advanced section enable the first checkmark.

SergioWarriorPeru, 2022-10-21:
Thanks...I can not pair with my Nuky 2.o....I have the same broad marked witd "X" in picture above swowed...it is frustrating for me

SergioWarriorPeru, 2022-10-21:
I have seen logs and it is always showing
Nuki start pairing
Nuki start pairing
Nuki start pairing
Nuki start pairing
Nuki start pairing
Nuki start pairing

technyon, 2022-10-22:
@gonzalezcalleja Do you have a link to the DUBEUYEW board? I'd like to add it to the readme as an incompatible board.

@SergioWarriorPeru Try a different board. It seems this one in particular has some issue.

SergioWarriorPeru, 2022-10-22:
Yeah...but, I would like know why...Maybe it could be the chip ESP32-D0WDQ6....I was surfing to know what is the chip used in chip of az-Delivery...but I have been successful

technyon, 2022-10-24:
Doing a quick google check, it's at least listed as "Not recommended for new designs", which means as far as I understand that there newer revisions available that address potential hardware bugs:

https://www.italki.com/en/teacher/7737037/englishhttps://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf

Maybe pick one that's not listed as "NRND".

gonzalezcalleja, 2022-10-24:
> @gonzalezcalleja Do you have a link to the DUBEUYEW board? I'd like to add it to the readme as an incompatible board.

Hi, I only have this amazon link: https://www.amazon.es/gp/product/B0B55MS33L

SergioWarriorPeru, 2022-10-25:
Yeah..I have the same DEVKITV1...I have just ordered a ESP32-WROOM-32D DEVKITV4 ...I will test when it arrive....

technyon, 2022-10-25:
@SergioWarriorPeru Please let me know if it works with the new ESP. If you can confirm it works, I'll mention DUBEUYEW  boards as not recommended in the readme.

SergioWarriorPeru, 2022-10-25:
@technyon ...Yeap..but, here (in Peru) orders from Ali Express can take 2 months..so i will be patient...I believe that problem is the version V1...

kouske, 2022-10-30:
I got the DevKitV1 board as well, can't pair the lock. I even have 2 of the same unfortunately..
Ordered one with ESP32-WROOM-32E, will report back if it works.

technyon, 2022-11-04:
Ok thanks, I'll add it to the readme.

bbo76, 2022-11-09:
Hello there, I just bought 3 Az Delivery ESP32 WROOM32. Trying on 2 different esp32 and the result is the same: loop on pairing and not found the smartlock.
I have a SmartLock V3. Nothing more with the debug version

bbo76, 2022-11-10:
> Oh, new logs:
> 
> ```
> [289304][D][NukiBle.cpp:102] pairNuki(): No nuki in pairing mode found
> [289305][D][NukiBle.cpp:107] pairNuki(): pairing result 0
> [294285][D][NukiBle.cpp:229] onResult(): Found nuki in pairing state:  addr: xx:xx:xx:xx:xx:xx
> Nuki start pairing
> lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
> [302732][W][NukiBle.cpp:138] connectBle(): BLE Connect failed, retrying
> [332746][W][NukiBle.cpp:138] connectBle(): BLE Connect failed, retrying
> [362759][W][NukiBle.cpp:138] connectBle(): BLE Connect failed, retrying
> ```
> 
> I already saw this error "lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)", it was related to the poor bluetooth connection. Did you max the bluetooth range power currently from the ESP32?
> 
> I cannot make the ESP32 closer to the smartlock, they are already together...

Did you solve the problem ? I have exactly the same scenario, sometimes I have nuki detected but BLE connect failed after

SergioWarriorPeru, 2022-11-10:
New ESP32 have arrived today and I have paired with Nuki (ESP32-WROOM-32D DEVKITV4)...but, I can not to integrate with Home Assistant...I have put homeassistant in Home Assistant discovery topic...but my HA can;t see it

Logs are:

Restarting
*wm:[2] [EVENT] WIFI_REASON:  8
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
Network device: Builtin WiFi
MQTT without TLS.
*wm:[1] AutoConnect 
*wm:[2] ESP32 event handler enabled 
*wm:[2] Setting Hostnames:  nukihub
*wm:[2] Setting WiFi hostname 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to SAVED AP: GWIFI
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] AutoConnect: SUCCESS 
*wm:[2] Connected in 1034 ms
*wm:[1] STA IP Address: 192.168.86.66
WiFi connected: 192.168.86.66
Host name: nukihub
MQTT Broker: 192.168.86.27:1883
NUKI Lock enabled
Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: no
NUKI Opener disabled
Presence detection timeout (ms): 60000
Attempting MQTT connection
MQTT: Connecting with user: sergio
MQTT connected
Nuki start pairing
Nuki paired
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
Unable to setup HASS. Invalid config received.
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
Reading config. Result: lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
3
Reading advanced config. Result: lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)

mironalessandro, 2022-11-11:
> New ESP32 have arrived today and I have paired with Nuki (ESP32-WROOM-32D DEVKITV4)...but, I can not to integrate with Home Assistant...I have put homeassistant in Home Assistant discovery [...]

I was thinking the same but if you go on entities you must find it if you set autodiscover.

SergioWarriorPeru, 2022-11-11:
@mironalessandro   I have looked to entities and I can't find something like nukihub or similar...
I do not understand why "Unable to setup HASS. Invalid config received"...I have reviewed Mqtt client and discovery topic is homeassistant the same I put in ESP32...

SergioWarriorPeru, 2022-11-11:
Finally, I could link ESP32 with Home Assistant...I had to install again and put homeassistant in "Home Assistant discovery topic" before to pair  the first time...I don't know why when Nuki was paired without homeassistant link and tried to configure it didn't work...

bbo76, 2022-11-11:
> Finally, I could link ESP32 with Home Assistant...I had to install again and put homeassistant in "Home Assistant discovery topic" before to pair the first time...I don't know why when Nuki was paired without homeassistant link and tried to configure it didn't work...

Hello Sergio, could you send exactly the model of the working board please

SergioWarriorPeru, 2022-11-11:
> Hello Sergio, could you send exactly the model of the working board please

Hi, yes...at the beginning I was trying with ESP-WROOM32...Now with ESP32-WROOM-32D DEVKITV4

yaronkof, 2022-11-11:
> Hi, yes...at the beginning I was trying with ESP-WROOM32...Now with ESP32-WROOM-32D DEVKITV4

Do you have a link?

SergioWarriorPeru, 2022-11-11:
Again, with DEVKITV1, for me, was not possible to pair with Nuki...but with my new ESP32-WROOM-32D DEVKITV4 it works good, at least lock, unlock and door state......Now I have to try open

> Do you have a link?

https://es.aliexpress.com/item/4000089201026.html?spm=a2g0o.order_list.0.0.21ef194db9a7f7&gatewayAdapt=glo2esp

technyon, 2022-11-12:
Hi,

Although it works now, the error "Unable to setup HASS. Invalid config received." means the configuration couldn't be retreived from the Lock, meaning something went wrong with the bluetooth communication ... not sure though why.

bbo76, 2022-11-13:
Hi,
Just reveived new board (with ESP32-WROOM-32D chip) and everything work like a charm. So I return my AZ delivery with WROOM-32, I think this chip is not compatible.
NB: My Az delivery board is devkit v2, it would be interesting to know if the devKit v4 is compatible

SergioWarriorPeru, 2022-11-13:
> Hi, Just reveived new board (with ESP32-WROOM-32D chip) and everything work like a charm. So I return my AZ delivery with WROOM-32, I think this chip is not compatible. NB: My Az delivery board is devkit v2, it would be interesting to know if the devKit v4 is compatible

Hi, yes, my ESP32-WROO.-32D is devkit V4 and work well...the only thing that I noted is when I open door in manual way, sometimes doorsensor does not shoew open but closed...I

kouske, 2022-11-13:
Well,  I ordered ESP32-WROOM-32D DevKit V4 and ESP32-WROOM-32E NodeMCU. 32D did not work at all, 32E did work.
It just may be that either these ESP32 SIP's are the issue or the board manufacturer is the issue, IDK anymore lol

amastrogiacomo1968, 2022-11-25:
Unfortunately I also have a DEVKITV1 that doesn't connect to the lock :-(
I'm ordering a new one, any suggestions? Thank you

Ps how can I recycle the unused DEVKITV1?

kouske, 2022-11-27:
I got the WROOM32-E and it seemed to work

technyon, 2022-11-28:
I've updated the readme to include known bad boards (devkit v1) and known working boards (32d and 32e).

davide93, 2022-11-29:
Az-delivery V2 (photos), not working for me. Some Bluetooth devices not found (Nuki too). I'm now waiting for the new version (V4) by Az-delivery nex days. Finger crossed.

Davide

[image]
[image]

gonzalezcalleja, 2022-12-01:
Hi @davide93,

Is V2:
[image]

davide93, 2022-12-01:
Really strange situation. Same version, but mine doesn't work and stays in the "Nuki start pairing" state. @technyon any advice?
@gonzalezcalleja  are you using the 3.4.10 firmware on Nuki 3.0?

Ty.

technyon, 2022-12-01:
Can anyone ship a non-working ESP to a german address? Best way to fix it if I can debug where the pairing gets stuck.

0rsa, 2022-12-02:
Morning all,
Finally I switched from Nuki 2.0 to Nuki 3.0 Pro, this is something I suggest to consider.
I suscribed to the nuki beta program (open to all, you just have to contact the Nuki support and they add you to the program within less than 24H)
With the beta firmware, the smartlock is able to communicate through a local MQTT server using built-in Wifi.
It's super fast compared to the bluetooth version (I'm now able to see the full workflow in real time in my personal app : locked -> **unlocking** -> unlocked in real time), thanks to push events.
You don't need to have a bridge anymore (official bridge or ESP32 DIY).
I will not go back on a bluetooth-based solution

massaquah, 2022-12-02:
> Morning all, Finally I switched from Nuki 2.0 to Nuki 3.0 Pro, this is something I suggest to consider. I suscribed to the nuki beta program (open to all, you just have to contact the Nuki support [...]

But within a combination of Nuki 3.0 (non-pro) and Nuki Opener you would still need a bridge because neither has built-in wifi, correct?

technyon, 2022-12-02:
Only the Nuki Pro has built-in WiFi, for anything also you need some kind of bridge (official, Nuki Hub, ...).

Let's please keep the conversation on-topic.

bogdal, 2022-12-02:
Hey 👋 I have tested multiple esp's with my nuki 2.0
- 🔴  `esp-s2` is not supported
- 🔴 `esp-c3` is not supported
- 🔴 `esp-wroom-32` (devkitv1) doesn't work (unable to pair lock)
- 🟢 `esp32-wroom-32UE` works (paired with lock)
- 🟢 `esp32-wroom-32E` works

davide93, 2022-12-02:
Hi there.
I got two more different "ESP" and the result is very strange and doesn't support the hypothesis that V1 doesn't work.

On my side:

🔴 **AZ-Delivery V2** doesn't work | [Amazon](https://amzn.eu/d/9xHWihg) | Photos: [card](https://user-images.githubusercontent.com/3228835/204643406-da0ba92c-3985-4b33-aaf2-40db60684150.jpg) - [box](https://user-images.githubusercontent.com/3228835/204643228-a8451d0e-37d6-4d93-9753-3cb994b62ee8.jpg) | [Product page](https://www.az-delivery.de/products/esp32-developmentboard?_pos=6&_sid=f2a44964f&_ss=r&variant=36542176914)

🔴 newest **AZ-Delivery V4** doesn't work | [Amazon](https://amzn.eu/d/9xHWihg) [Amazon link](https://amzn.eu/d/az4qOMQ) | Photos: [card](https://i.ibb.co/q1XYTbB/IMG-20221202-210609.jpg) - [box](https://i.ibb.co/wwwBmGL/IMG-20221202-213217.jpg) | [Product page](https://www.az-delivery.de/products/esp32-dev-kitc-v4-ai-thinker-soldered?_pos=23&_sid=f2a44964f&_ss=r)

🟢 the oldest  **JZK "V1"** (which requires the "boot" button to be pressed when uploading code) worked right away | [Amazon](https://amzn.eu/d/16PevXN) | Photos: [card](https://i.ibb.co/pw4qzY8/IMG-20221202-210651.jpg) - [box](https://i.ibb.co/QksfDY4/IMG-20221202-211106.jpg) | [Product page](https://www.jzk-jzk.com/collections/arduino-supplies-1/products/esp-32s-esp32-development-board-2-4ghz-wifi-bluetooth-dual-mode-antenna-module-with-ultra-low-power)

(I may be wrong, but JZK resells DOIT brand products even though neither the JZK nor DOIT brand is listed on the board.)

... but someone has confirmed that AZ-Delivery V2 works.
I almost think there are defective batches? I was surprised that even the new  v. 4 did not work and instead v. 1 worked right away.

@technyon Forgive me but right now I am not able personally to send one of these to you. Anyway, thank you for your amazin work.

Stilzchen, 2022-12-05:
Hi,

I tested 3 different ESPs, non is working:

- Espressif ESP-WROOM-32, Chip is ESP32-D0WD-V3 (revision 3)
   Labeled with "ESP32S" at bottom
   [Link](https://www.amazon.de/gp/product/B07S1GGM35/ref=ppx_yo_dt_b_asin_title_o07_s01?ie=UTF8&psc=1)

- AZ Delivery ESP-WROOM-32, Chip is ESP32-D0WD-V3 (revision 3)
   No other labels at board but box was labeled with "DevKitC V2" like [here](https://github.com/technyon/nuki_hub/issues/39#issuecomment-1333490521)
   [Link](https://www.amazon.de/gp/product/B071P98VTG/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1)

- Espressif ESP32-WROOM-32D, Chip is ESP32-D0WDQ6 (revision 1)
   Labeled with "ESP32_DevKitC_V4" at bottom.
   [Link](https://www.amazon.de/gp/product/B09LS6SCF7/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

Non of them is working with my Nuki 3.0
Tested with nukihub 6.0, 6.0-dbg-1, 6.1 and 5.8, 5.15 and a very old 2.6 release.
Tested with "register as app" on and off.
Restored to factory settings the Nuki smart lock.

@technyon: I can send some devices, if needed

Next I will try on of the posted links here or a 32U

[43 more replies]

closed 2023-04-04 by technyon