technyon/nuki_hub #40: add 2 or more locks

sconvolto23, 2022-10-09:
hi folks,

It's possible handle 2 locks with only one board esp32?

Mincka, 2022-10-10:
I tried to put my second lock in pairing mode, while the first one was already paired with `nuki_hub` identified as an app, but no luck.
I think the `nuki_hub` is not open to pairing once one device is paired. Also, I am not sure that the mobile itself it simultaneously connected to both locks in BT.
Indeed, it would be awesome to support multiple lock with the same ESP32 board. 🔥
Thanks for this awesome project.

technyon, 2022-10-10:
Hi and sorry for the late answer. It's not possible to pair more than one lock (or opener) at this time ... not sure how I'd handle that in the UI. For now just use another ESP, shouldn't be a problem except maybe power supply for two devices.

kvj, 2022-10-11:
+1 to what @technyon said. ESP program will always be single-threaded, speaking of IO and even if theoretically it's possible to maintain credentials and re-organize UI for multiple devices, performance and reliability -wise it won't be the right approach. 

You can buy another ESP32 board for less than 7 Euro I think

Mincka, 2022-10-11:
Thanks for the feedback.
I'll be watching if they manage to handle this in this other project:
https://github.com/uriyacovy/ESPHome_nuki_lock/issues/2

closed 2022-11-28 by technyon

bcutter, 2023-01-15:
It's really just because of "3 smart locks equals 3 ESPs equals 3 occupied plug sockets and 3 times power usage".
For all the other points made (especially complexity and performance) I understand why it has no priority and has been closed.

For having up to 5 Smart Locks paired the official Nuki Bridge seems to be the way to go. Monetary speaking, with roughly 10 € per ESP including 3D printed case compared to discounted Nuki Bridge (~ 80 €) that makes a break-even of 8 Smart Locks - without taking power usage into account ;-)

kvj, 2023-01-16:
> For having up to 5 Smart Locks paired the official Nuki Bridge seems to be the way to go

5 locks within Bluetooth reach of a single bridge? Some Airbnb like setup maybe. The majority (number of users, not devices) of Nuki customers probably have one or two of those, installed in a single household. 

I doubt that Nuki bridge can maintain 5 simultaneous BT connections without issues, and I highly doubt that ESP32 can do the same

technyon, 2023-01-16:
I think power consumption of an ESP is minimal, the power supply is the bigger factor I guess. You could get a charger brick with multiple USB ports to minimize that, instead of heaving 5 seperate chargers. I do see how it would be convenient to only have a single device, but it my capacity both in coding and testing is limited. Also I only have one lock, so no way to properly test it.