technyon/nuki_hub #42: Keypad Control: when enabled Nuki Hub doesn't work anymore

alexdelprete, 2022-10-18:
Hi,

I just configured Nuki Hub: flashed fw, configured wifi, set mqtt parameters and paired the Lock. Automagically HA had all the entities available, they were refreshing quickly. Everything worked out beautifully.

While checking the other features, I enabled Keypad Control: the device rebooter, and NOTHING was working anymore. Ping timeouts on the network, ping timings were huge, and then it stopped responding to pings. I cyvled power a couple of times but the behaviour was consistent, after the boot, things weren't working anymore, the device was not reachable. Luckily, I managed to open the web interface for 10 secs one time and I could disable the Keypad Control, the device rebooted and everything was perfect again.

While it wasn't working, I noticed a LOT of MQTT msgs on the nuki/keypad topic. I think the CPU of the device was in a loop, that's why nothing was working anymore (that was the impression at least).

Hope you'll find the issue. Meanwhile, thanks a lot for this project.

alexdelprete, 2022-10-20:
> It sounds like your ESP was stuck in a reboot-loop

If it is a reboot loop, it's due to the enablement of that Keypad option. As soon as I turned it off, everything was good again. If I turn it on again, I'm back at square one. And also, the quantity of messages received by the broker was definitely abnormal.

From what I experienced, I think it's a sw issue (maybe specific to my configuration?) that causes the device to fire a lot of messages, and probably the CPU goes crazy and the device gets unresponsive. But I'm just speculating.

> The keypad is configured and has some codes set?

Yes it is. We don't use it often, but it is configured.

> Which MQTT broker are you using?

I use EMQX v5.0.8.

technyon, 2022-10-20:
What is set as "Query interval keypad" in your ESP?

alexdelprete, 2022-10-20:
[image]

polysa, 2022-10-27:
I have the same problem after enabling the Keypad from the Dashboard. 

MQTT is still working, and I can use the lock from Home Assistant. How can I disable the keypad without access to the dashboard?

thanks

bcutter, 2022-12-29:
Progress?

technyon, 2022-12-30:
It's really hard to do anything about this if I can't reproduce the problem. Plus it's one of those things where sending some binary doesn't really help, I'd need access to an ESP with that problem to be able to debug. I've reviewed the code several times but didn't find anything that would trigger these reboots.

knorrre, 2023-01-28:
Hello,
is this issue the reason why the option "Enabled keyboard control via MQTT" is currently not displayed in the WebGUI? At least this is the case for me:
https://i.imgur.com/54suCV5.png

technyon, 2023-01-28:
No :). Are you by any chance trying to use it with the Opener? That's not supported yet.

knorrre, 2023-01-28:
No, it is about the Keypad 2.0. In the WebGUI I am not offered the possibility of activation via MQTT, see screenshot.

technyon, 2023-01-28:
Interesting ... I have no 2.0 so I can't check. But I guess it should work, since there weren't any changes to the API. As soon as it's connected to the lock, you should have the option to enable keypad control.

knorrre, 2023-01-28:
Weird! :)
Door lock and keypad are successfully connected and working for a few days. But apparently the option does not exist. Do I need to set something else somewhere?

technyon, 2023-01-28:
No. Maybe 2.0 does work differently from the old keypad. The logic is very simple ... the lock reports is has a keypad via a flag in the config. If the flag is set, you get the option.

knorrre, 2023-01-28:
Well, the entity of the keypad battery is visible in Home Assistant. Is it possible to set the flag or the option by myself?

technyon, 2023-01-28:
That is actually a bug ... the entity shouldn't be there. The binary below overrides the flag, but most likely it won't help ... unless they changed the way 2.0 is detected and didn't document it yet. Let me know if you can add/remove key codes, in that case we can ask in the Nuki developer forum.

[nuki_hub-keypad-override.zip](https://github.com/technyon/nuki_hub/files/10527154/nuki_hub-keypad-override.zip)

knorrre, 2023-01-28:
The option is now visible after installing the firmware and the check mark could be set. Messages "keypad" are visible via MQTT. Thank you!
To be frank, I do not know exactly how to add or remove codes now. About the app are in any case four codes created, but which are not shown via MQTT - if I understand correctly. Here is a screenshot:
https://imgur.com/a/kOvD8NZ

technyon, 2023-01-28:
If this works there's would be nodes creates for every code you've setup in the app, just numbered from "0" to the number of codes you have. If you don't have those, overriding the flag won't help you.

knorrre, 2023-01-28:
There are four codes created, but none (of them) is output. So you are right, overwriting the flag does not help. I am at a loss. What now?

technyon, 2023-01-28:
That's good news, it does work. The codes themselves aren't output due to security reasons. You can modify them though via the command-nodes.

technyon, 2023-01-28:
P.S.: Sorry I misread. Four codes you mean in the app or you see them in the MQTT tree?

knorrre, 2023-01-28:
I do not think this is good news, rather that I have misspoken. :)
Four codes were set up via the Nuki app, but only one code is issued via MQTT (see screenshot above), which doesn't even have a name.

alexdelprete, 2023-01-29:
I tested thia again, with v8: finally it seems to not create that issue of OP. :)

But I don't see the 3 codes I registered for the keypad:

[image]

knorrre, 2023-01-29:
On top of that, I can no longer unlock or lock the door via Home Assistant's UI. (However, I am unsure whether this worked before. I've only had the lock for a short time).

alexdelprete, 2023-02-05:
@technyon Jan, should we keep this open or do we close it for now? The OP issue is solved, there's just a functional issue regarding the codes.

technyon, 2023-02-06:
Let's close it and create a seperate issue if necessary

closed 2023-02-06 by technyon