technyon/nuki_hub #43: esp32 haltect lora32 v2 board

drdino, 2022-10-31:
Hello,

I have heltec esp32 board with lora32 v2 wifi module. Flashing was successfull but when i flash with nuki hub firmware i got console message that ethernet cannot be found. Is it possible to make support for lora32 as well? Here is the board description: https://heltec.org/project/wifi-lora-32/

thanks

technyon, 2022-11-04:
If I understand correctly the ESP is trying to connect via ethernet and not opening the Wifi Manager config portal? Looking at the datasheet, Pin 26 is connected to "LoRa_DIO0", which probably switches the firmware to W5500 ethernet mode - when Pin 26 is pulled to ground it disables WiFi and enables W5500.

 Not so sure how to fix that, I had to pick some GPIO to activate ethernet. Choosing a different PIN probably makes it incompatible with other boards.

technyon, 2023-01-01:
I finally found a solution that should enable this device to work. If no network hardware is found, the ESP reboots and falls back to Wifi. Once in Wifi fallback mode, you can configure the device to disable W5500 in the configuration portal (under MQTT and network configuration)

closed 2023-01-08 by technyon