technyon/nuki_hub #44: MQTT vs GPIO (GPIO seems unstable and slower to respond)

tongaimaramba, 2022-11-02:
I'm controlling the nuki hub via home assistant (which is installed on an RPi 4). On the (maybe mistaken) assumption that a direct gpio connection will have faster and more reliable response times, I've enabled gpio control. But I find that the response times are very similar. Also, the lock often fails to respond via gpio whereas via mqtt it always responds.

I notice that with mqtt I can send another command as soon as the last one completes. But with GPIO I usually get no response if I do that. Does the GPIO implementation have a minimum time between commands?

I haven't been able to find any logging for nuki hub, where would I find that?

Separately - Bluetooth: how does distance between nuki hub and lock affect response time? To co-locate the nuki hub with home assistant, the nuki hub is about 4m away from the lock with a brick wall in between. Note: for the MQTT vs GPIO tests above, the nuki hub was colocated with homeassistant. So 

Thanks.

technyon, 2022-11-04:
Hi,

The problem is not the GPIO itself, but the lock. It takes a certain time for a command to complete, if you send commands too fast you'll just get a "busy" error back from the lack. This can even happen when it the lock has stopped moving and appears to be finished. In MQTT you get the command result code, so you know when it's finished and you can only then send the next command. On GPIO this information is not available, but usually this shouldn't matter because the normal use case isn't rapid locking and unlocking.

I don't have any controlled tests regarding WiFi and bluetooth reception. Place it as close as you can and test, move it around if necessary. The PCB antenna on most ESP boards isn't great, it might be a good idea to buy an ESP with an external antenna.

tongaimaramba, 2022-11-04:
Thank you, this is helpful. 

I was trying to understand the performance tradeoffs between  (a) having the nuki hub close to the lock but then driven over mqtt; vs (b) having the nuki hub directly wired into homeassistant (via GPIO) but then separated by a greater distance from the lock.

Given the better exception handling of mqtt, I think option (a) is the way to go.

Many thanks!

closed 2022-11-28 by technyon