technyon/nuki_hub #45: Integrated wiegand Keypad

uvs1313de, 2022-11-02:
It works perfekt and stable. I used one more ESP32 with wiegand interface to use keypad and Rfid. One output pin connected to the unlatch pin of your nuki hub. It was a first version for testing and it works. So I could use Fingerprint Keypad and your nuki hub. 
Now I am thinking of integrating these wiegand interface code into your nuki hub code as a all in one solution.

technyon, 2022-11-04:
Hi,

I don't really have experience with the wiegand protocol, but there are some libraries available for the Arduino - they might need some tweaking to get them working on the ESP. Plus I guess you'd need a level shifter as wiegand runs on 5V.

uvs1313de, 2022-11-04:
Hi, thank you for the quick reply. Right. I created the sketch (first very simple version) for wiegand on ESP32 using some of the existing libraries and as you wrote "tweaking" to get them work. I used Pin 4 and 5 for Data0 and Data1 of the wiegand interface and level shifter to convert 5V to 3.3V. At the first version i just hardcoded the keypad and Rfid number in the sketch. Just for testing. Pin 16 is now connected as Output to another ESP32 running with your code for Nuki Hub and there to the Pin for "unlatch". This is works fine. Now i am trying to combining it with your code, if its ok for you, to use only one ESP32 and use the web interface to maintain even the keypad numbers (create new or delete existing).

technyon, 2022-11-05:
Hi,

go ahead and add it. For now I'd hardcode the GPIOs, but I see the need to make the GPIOs configurable sooner or later. There's already a request to make it compatible with a certain board which uses GPIO 26 which kicks the firmware into ethernet mode, thus no Wifi connection is possible. I'll see to add something to the web interface, but we can do that in a later step.

uvs1313de, 2022-11-07:
Hi,
you are right. Will add this to disable this in web interface on my list. At the moment I am struggling a bit with cmake and the arduino tool chain to use your source code. Normally I use Arduino IDE and platformio. In parallel I go through your code to understand where to put the code for wiegand in and how to design the HTML. My first idea was to place the wiegand code in loop in main.cpp as a first test. To build the bin files I then need cmake. So still a lot of things to do 😊

technyon, 2022-11-07:
For a howto on the toolchain check my other project:

https://github.com/technyon/coffeetimer

Also, you have to copy the partitions.csv into the cmake directory:

[partitions.csv](https://github.com/technyon/nuki_hub/files/9950068/partitions.csv)

This project uses FreeRTOS, so you can't use the main loop. Instead create a new task (see setupTasks() in main.cpp). Make sure to put a delay into the tasks loop, otherwise you'll trigger the watchdog and get a reboot.

Setting up the web server is quite straightforward, check the option to enable or disable GPIO. It's two places where you have to add code, one to display the option, the other one to check wether it has changed or not.

uvs1313de, 2022-11-07:
Hi,
thank you for the quick answer :-)

technyon, 2022-11-28:
Did you make any progress on this, or need any help?

uvs1313de, 2022-11-29:
Well, I started to rewrite my code for rtos but have still some interrupt and timing problems.
In parallel I started to create web interface to maintain the pin codes.
At the moment I am using a second Esp32 for wiegand and connected it on pin 27 of another ESP with your code. I think it takes a bit more time to get the wiegand code for rtos working.

technyon, 2022-11-29:
ok thanks for the update

technyon, 2023-02-05:
Hey, any update on this.

Feel free to join us on discord:

https://discord.gg/feB9FnMY

uvs1313de, 2023-02-07:
Hi, unfortunately not because of less time. I was working on epaper display as information panel. 
But Esp32 hub and wiegand is still on my todo list 😊 I rewrote my wiegand code a bit and at the moment use a nuki fob which needs only 5 seconds to open the door. That was at the moment the fastest way. I am still struggling a bit with rtos and the way of programming. But everything starts with a first step 😊

I will have a look at disfhord

iranl, 2024-05-17:
I'll close this as:
- Topic has been stale for over a year
- Recent view has been to not include code for external devices/integration that are not directly linked to the functioning of Nuki devices. Nuki Hub allows allows access to almost any function of Nuki Hub through MQTT/GPIO from a second device/other application that deals with the wanted integration (A second ESP8266/ESP32 that deals that manages the Wiegand in this example)
- Nuki Hub is starting to approach the maximum available flash for 4MB devices. Adding libraries/code for external devices/integrations will add additional difficulties to remain within the available flash.

Please reopen if you disagree.

closed 2024-05-17 by iranl