technyon/nuki_hub #46: Home Assistant Support

jjprz, 2022-11-10:
Hello.

I am trying to configure my Nuki device. I have done the following steps: I have connected it to the WiFi and I have paired it. I just have to connect it to Home Assistant. But I have never used mqtt and I don't know how to connect it.

The console at https://technyon.github.io/nuki_hub/ shows me the following messages:

Attempting MQTT connection
MQTT: Connecting without credentials
MQTT connect failed, rc=-2
Free Heap: 116552
Nuki start pairing
Nuki paired
Attempting MQTT connection
MQTT: Connecting without credentials
MQTT connect failed, rc=-2
Free Heap: 114892
Attempting MQTT connection
MQTT: Connecting without credentials
MQTT connect failed, rc=-2
Free Heap: 113956
...

I also have MQTT enabled in Home Assistant.
[image]

What else should I do?

Thanks in advance.

technyon, 2022-11-12:
It seems you didn't configure MQTT correctly. Did you configure the correct address / port for the MQTT broker, and entered the credentials (if needed)?

kouske, 2022-11-14:
Worth mentioning that the username/password should be the ones that you added to homeassistant via Settings -> People -> Users.
Also, note that there is an autodiscovery section in the NUKI bridge MQTT configuration - set that to homeassistant (as technyon recommended).

@technyon , it took me a while to figure this out as I am a complete noob at this so may be worth adding to the README :)

jjprz, 2022-11-14:
I have configured mqtt with username and password in Home Assistant
The nuki_hub configuration process did not ask me for a username and password, so I do not know where to enter these credentials.

mundschenk-at, 2022-11-14:
You need to set up under "MQTT and Network Configuration" (`/mqttconfig`).

technyon, 2022-11-29:
Does it work now?

technyon, 2022-12-04:
Closing due to inactivity. Let me know if issue persists, so I can reopen.

closed 2022-12-04 by technyon