technyon/nuki_hub #47: Flashing suceed but ESP32 in constant reboot loop

milansim, 2022-11-15:
Device chinese copy of "Wemos D1 mini ESP32"
Previously worked fine with arduino firmware with OTA, wifi, MQTT&Home assistent...Then I wanted to try your hub with my lock

Tried erase and then flash trough edge browser, also with espresif tool, all went fine. 
After flashing I cant find it's access point and here is a log from your web flashing tool 

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)

Backtrace:0x40083b79:0x3ffe3aa00x4009314d:0x3ffe3ac0 0x40098b65:0x3ffe3ae0 0x401305a2:0x3ffe3c10 0x4008327a:0x3ffe3c40 0x4007920a:0x3ffe3c90  |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0

assert failed: do_core_init startup.c:298 (flash_ret == ESP_OK)

Backtrace:0x40083b79:0x3ffe3aa00x4009314d:0x3ffe3ac0 0x40098b65:0x3ffe3ae0 0x401305a2:0x3ffe3c10 0x4008327a:0x3ffe3c40 0x4007920a:0x3ffe3c90  |<-CORRUPTED 
...

technyon, 2022-11-16:
I'd say something went wrong with flashing the firmware. Please try again with the espressif download tool, and follow the included readme in the firmware .7z archive exactly. Other than that, you could try to factory reset your ESP before flashing.

milansim, 2022-11-16:
Seems some problem with my ESP32 board...cant use official tools to erase (but arduino can ?)
I asked a friend with same board and he flashed no problem, device work fine, so we will swap boards, problem solved :-)

technyon, 2022-11-16:
Considering how cheap ESPs are, probably the easiest solution. I'll close the issue, reopen it if you still have problems.

closed 2022-11-16 by technyon