technyon/nuki_hub #48: CSS load issue

adrienvsj, 2022-11-18:
Hello,

Every other time I can't load the CSS of the pages and I get this error in the console : 

`http://192.168.1.26/new.css net::ERR_CONNECTION_RESET`

I have to refresh the page to get the display back.
I use this ESP32: https://www.amazon.fr/gp/product/B08BTLYSTM
I use firmware 6.0

Thank you

technyon, 2022-11-28:
Hi and sorry for the late answer. I don't have an idea really, it works for me. Does the "inter.css" load?

technyon, 2022-11-30:
P.S.: I'm running on wired ethernet. I'll try switching to WiFi and see if I can reproduce the issue.

alexdelprete, 2022-11-30:
I confirm the issue. I'm on wired ethernet with my notebook, but I don't think that how you connect to the network should influence the functionality, you can have more latency, but it should be the same, functionality-wise.

Also, refreshing doesn't help much, it seems random: if I refresh 4-5 times, eventually CSS pulls up, but only one time and in between I see connection reset or the page without CSS.

technyon, 2022-12-01:
I mean I'm on wired ethernet with my ESP, using a W5500 module.

alexdelprete, 2022-12-01:
Oh sorry, I assumed you had a wireless module like me.

technyon, 2022-12-03:
Hi,

I've switched to Wifi for testing, but couldn't reproduce the issue. I'd say it must be something on your end, although it's strange it's always this one file. Can you try with a different browser / computer ?

alexdelprete, 2022-12-03:
I tried with Chrome and Edge. Now I installed Brave on another computer:

[image]

alexdelprete, 2022-12-03:
I just installed Firefox: it always picks up CSS correctly. So it's something with chrome-based browsers. I don't know if Brave is chrome-based, will check it up. In any case, it should work on all browsers.

[image]

technyon, 2022-12-04:
I tried with both Firefox and Chrome and had no issues. Maybe do a clean install of chrome?

ruimarinho, 2023-01-02:
I'm also having the same issue on a fresh install of 6.8.

Frank-1964, 2023-01-24:
Same issue here with nuki_hub 7.2 an Chrome 108.0.5359.71 under Linux Ubuntu.

alexdelprete, 2023-01-24:
Something's not right with the webpage, here's what Edge (chrome based) is reporting:

[image]

The only browser I don't have problems with is Firefox, on ALL chrome-based browsers there's a compatibility issue.

technyon, 2023-01-25:
I'm not a web developer, if someone wants to fix the HTML go ahead :D. The page displays fine except for some users where the CSS for some reason doesn't load. I have no explanation for that, it's static content that's delivered if requested. What's even stranger is that inter.css loads, but new.css doesn't - it's pretty much the same code.

The web page is still usable - it's a simple table based page that is just looks a bit nicer with the CSS.

alexdelprete, 2023-01-25:
> The web page is still usable - it's a simple table based page that is just looks a bit nicer with the CSS.

No it isn't. It takes a lot of time to load, and sometimes it throws errors (RESET, timeout, etc.). I installed Firefox just to use Nuki Hub. :)

I'm not a dev, but I'll check if there are some validation tools online that maybe will optimize the CSS code.

technyon, 2023-01-25:
WebCfgServerConstants.h lines 6 and 10. You'll find the URL where I copied it from in the comments above.

alexdelprete, 2023-01-25:
> P.S.: Still no issue for me, no matter which browser I use.

These things drive me crazy. Many of us have issues with that...on Chrome. How is it possible you have no problems...:)

mundschenk-at, 2023-01-25:
Maybe a different OS/version?

alexdelprete, 2023-01-25:
> Maybe a different OS/version?

We found the problem...and fixed it. Great teamwork on Discord. Join us. :)

technyon, 2023-01-26:
Please check release 7.3

regevbr, 2023-01-26:
> 7.3 works for me now. Thanks :-)

Same here many thanks!

technyon, 2023-01-26:
ok I'll close the issue then

closed 2023-01-26 by technyon