technyon/nuki_hub #49: Nuki Opener Ring doesnt get recognized everytime

G3rtz, 2022-11-24:
Hello everyone,
im testing nuki_hub as an alternative to nuki_ng, which always seems to drop connection for a few seconds, so i get 5-10 times the homekit announcement, that my door is now shut / opener is off.

i have it paired as app at the moment, because of the Wife Acceptance Factor. The Bridge is annoying with the messages, but if it doesn't work, would be much worse.

Nuki Hub seems to not register all the events, that are going on.
Here is a screen of the states in HA
[image]

And here is a MQTT Debug Log
[image]

So, what am i doing wrong?

technyon, 2022-11-24:
Can you give some more information? As a far as I can see from the debug output, there were two rings detected ... times don't match up though with the picture above. Which ring wasn't detected?

G3rtz, 2022-11-24:
sorry, i wasn't really clear with it. The MQTT Log corresponds to the upper bar (its the "Haustuer Summer ring", not "ring action")
"Nuki Haustuer Summer ring action" is from the Nuki Bridge via nuki-ng. The ring at ~8 AM and 9:15AM are both missing.

technyon, 2022-11-28:
I'm not really sure what could be the problem. I had to take a guess how the opener signals a ring, since this is completely undocumented. From what I could guess, whenever the opener signals a change to "locked" when the state is already "locked", a ring has been detected. Could you observe in your MQTT logs this is the case? Was the opener in the "locked" state when the ring occured?

G3rtz, 2022-11-29:
i dug into yesterdays log, as there were two ring events happening.
[image]
this is from nuki via nuki bridge and nuki-ng. as you can see, there was a ring event.

[image]
this is from nuki-hub. there was no event. The binaryState is nondescript also:
[image]

Despite everything, the lock state seems to have the event, but it seems to not be submitted to Home Assistant:
[image]

technyon, 2022-11-29:
It's hard to tell because as I mentioned, the ring detection is undocumented. You can try the attached firmware for debugging, it outputs the previous and new state of the opener on every change via serial. Whenever you see this, a ring is detected (and should be reported via MQTT):

Opener state updated: locked --> locked

Maybe you can figure out if any other state changes should report a ring.

[nuki_hub.zip](https://github.com/technyon/nuki_hub/files/10115288/nuki_hub.zip)

iranl, 2024-03-02:
With the changes in 8.33 a ring that is visible in nukiopener/lock/state should now always signal a Home Assistant ring in the new ring event that is made available through auto discovery.

Can be closed.

technyon, 2024-03-03:
Closing for now.

@G3rtz Can you recheck if release 8.33 fixes your problems. If not, we'll reopen the issue.

closed 2024-03-03 by technyon