technyon/nuki_hub #50: mqtt restart command

adizarko123, 2022-11-26:
Hi,
Great project!
Everything is working great, but sometimes the esp32 is losing connectivity.

May I know what the MQTT restart command that I will have a failsafe process for these kinds of problems is?

mundschenk-at, 2022-11-26:
Do you know whether you are losing WiFi or BT connectivity? I've identified an issue with BT connectivity being lost between Nuki and the ESP32, my HA automation workaround is documented in https://github.com/technyon/nuki_hub/issues/28#issuecomment-1272508865. You will also find the MQTT restart command there (but that will obviously only help if the issue is with BT connectivity).

adizarko123, 2022-11-28:
Thank you for sharing the mqtt command.
I'm secpecting it is related to my network as I have also zigbee2mqtt that also lose the mqtt connectivity on the same time.
I did some failsafe process also for the docker that is running my unifi key but would like to have also one for the bridge as it is security component in my house.

mundschenk-at, 2022-11-28:
@adizarko123 If your ESP32 loses the connection to the MQTT server, you can schedule a reboot via [firmware configuration](https://github.com/technyon/nuki_hub#random-wifi-disconnects). Indeed this would be the only way because the device would not receive your reset topic if it is not connected to the server anymore.

adizarko123, 2022-11-28:
@technyon - thank you for you advise.
The last time it happened, a restart command helped.

I will keep monitor it.

technyon, 2022-12-24:
Hi ... Closing due to inactivity, I guess the issue is fixed. Let me know if not we can reopen the issue.

closed 2022-12-24 by technyon