technyon/nuki_hub #53: Slow/no response for lock/unlock action over MQTT

giejay, 2022-12-02:
Im using this project together with HA. Whenever I try to unlock/lock both of my doors (have 2 Nukis and 2 AZDelivery ESP32, https://www.amazon.nl/dp/B071P98VTG/ref=pe_28126711_487102941_TE_SCE_dp_1?th=1), I get a 30/70 percent success rate.

Success:
```
front-door/lock/action unlock
front-door/maintenance/wifiRssi -33
front-door/lock/rssi -64
front-door/lock/action ack
front-door/maintenance/wifiRssi -34
front-door/presence/devices 54:d2:72:xx:1a:xx;Nuki_xxxx;-65
front-door/lock/commandResult success
front-door/lock/rssi -65
front-door/maintenance/wifiRssi -33
```

No ack/commandResult:
```
front-door/lock/action unlock
front-door/maintenance/wifiRssi -33
front-door/presence/devices 54:d2:72:xx:1a:xx;Nuki_xxxx;-76
front-door/maintenance/wifiRssi -33
front-door/maintenance/wifiRssi -32
front-door/maintenance/wifiRssi -32
```

Simply nothing happens. Any idea how to debug this?

giejay, 2022-12-02:
Discovered something odd, I have reset my esp for the front door and flashed an esphome Nuki lib to restore it to my previous state. That one works again but now suddenly, my shed door also started working again using this library! 

It seems the two are interfering somehow but i am using different paths/prefixes for the topics. For now I will keep it like this and monitor my shed door the coming days. If it's all stable I will reflash my front door with this library again.

technyon, 2022-12-02:
Ok I hope it works ... if not let me know. Are your ESPs located in a way that they have both good Wifi and bluetooth reception?

giejay, 2022-12-02:
Yes, my front door is 20 cm away from the router and 1 meter away from the door. Shed door is 40cm away from the door, a bit further from the wifi but stable.

technyon, 2022-12-04:
Can you check and post the content of the HA autodiscovery topic? Maybe something is wrong in there, like the path not being replaced with the actual path from MQTT Settings.

giejay, 2022-12-04:
Sure: 
```
homeassistant/lock/xxxx1d6c/smartlock/config {"dev":{"ids":["nuki_xxxx1d6c"],"mf":"Nuki","mdl":"SmartLock","name":"Schuur"},"~":"shed-door","name":"Schuur","unique_id":"xxxx1d6c_lock","cmd_t":"~/lock/action","pl_lock":"lock","pl_unlk":"unlock","pl_open":"unlatch","stat_t":"~/lock/binaryState","stat_locked":"locked","stat_unlocked":"unlocked","opt":"false"}
homeassistant/binary_sensor/xxxx1d6c/battery_low/config {"dev":{"ids":["nuki_xxxx1d6c"],"mf":"Nuki","mdl":"SmartLock","name":"Schuur"},"~":"shed-door","name":"Schuur battery low","unique_id":"xxxx1d6c_battery_low","dev_cla":"battery","ent_cat":"diagnostic","pl_off":"0","pl_on":"1","stat_t":"~/battery/critical"}
homeassistant/sensor/xxxx1d6c/battery_voltage/config {"dev":{"ids":["nuki_xxxx1d6c"],"mf":"Nuki","mdl":"SmartLock","name":"Schuur"},"~":"shed-door","name":"Schuur battery voltage","unique_id":"xxxx1d6c_battery_voltage","dev_cla":"voltage","ent_cat":"diagnostic","stat_t":"~/battery/voltage","state_cla":"measurement","unit_of_meas":"V"}
homeassistant/sensor/xxxx1d6c/trigger/config {"dev":{"ids":["nuki_xxxx1d6c"],"mf":"Nuki","mdl":"SmartLock","name":"Schuur"},"~":"shed-door","name":"Schuur trigger","unique_id":"xxxx1d6c_trigger","ent_cat":"diagnostic","stat_t":"~/lock/trigger","enabled_by_default":true}
homeassistant/sensor/xxxx1d6c/battery_level/config {"dev":{"ids":["nuki_xxxx1d6c"],"mf":"Nuki","mdl":"SmartLock","name":"Schuur"},"~":"shed-door","name":"Schuur battery level","unique_id":"xxxx1d6c_battery_level","dev_cla":"battery","ent_cat":"diagnostic","stat_t":"~/battery/level","state_cla":"measurement","unit_of_meas":"%"}
homeassistant/binary_sensor/xxxx1d6c/door_sensor/config {"dev":{"ids":["nuki_xxxx1d6c"],"mf":"Nuki","mdl":"SmartLock","name":"Schuur"},"~":"shed-door","name":"Schuur door sensor","unique_id":"xxxx1d6c_door_sensor","dev_cla":"door","stat_t":"~/lock/doorSensorState","pl_off":"doorClosed","pl_on":"doorOpened","pl_not_avail":"unavailable"}
homeassistant/sensor/xxxx1d6c/wifi_signal_strength/config {"dev":{"ids":["nuki_xxxx1d6c"],"mf":"Nuki","mdl":"SmartLock","name":"Schuur"},"~":"shed-door","name":"Schuur WiFi signal strength","unique_id":"xxxx1d6c_wifi_signal_strength","dev_cla":"signal_strength","ent_cat":"diagnostic","stat_t":"~/maintenance/wifiRssi","state_cla":"measurement","unit_of_meas":"db"}
homeassistant/sensor/xxxx1d6c/bluetooth_signal_strength/config {"dev":{"ids":["nuki_xxxx1d6c"],"mf":"Nuki","mdl":"SmartLock","name":"Schuur"},"~":"shed-door","name":"Schuur bluetooth signal strength","unique_id":"xxxx1d6c_bluetooth_signal_strength","dev_cla":"signal_strength","ent_cat":"diagnostic","stat_t":"~/lock/rssi","state_cla":"measurement","unit_of_meas":"db"}

```

technyon, 2022-12-04:
Looks fine I think.

Did you try to set the MQTT nodes directly without using HA? For example, write "lock" or "unlock" to "lock/action". If it works like that, it's something related to HA.

giejay, 2022-12-05:
I have reflashed it but this time disabled the HA integration/auto discovery. I unlocked and locked the frontdoor a couple of times but noticed that I didnt get any feedback in the /state or /binaryState topic. The lock/unlock action was succesful however.

Then, I switched to the shed door and that one immediately stopped responding and gave the same issue as described in the first post. Not giving any ack commands and not locking the door. 

Lastly, I simply cut the power to the front-door ESP32 and yes, the shed door was fully functioning again. 

Only thing I can think of is now setting up a completely new broker (mosquitto in my case) :)

closed 2022-12-11 by technyon