technyon/nuki_hub #54: 2 nuki smart lock in the same house

pilot1981, 2022-12-07:
Hi,
I have a Nuki smart lock 3.0 (only BLE) and a Nuki smart lock 3.0 pro (BLE+WIFI) in the same house. they are not accesible via BLE from an unique point.

I can usa only one ESP32? it can connect via BLE near to Smart Lock 3.0 and via WIFI to the Smart lock 3.0 Pro?

or I have to use n.2 ESP32? one dedicated for each Smart lock?

regards

technyon, 2022-12-08:
Hi,

Each ESP can only control one lock and one opener, so you need two ESPs to control two locks.

You can only connect via bluetooth to the lock, but if you have A Pro lock, you can eventually connect it directly without the need of an ESP with all advantages and disadvantages.

Considering how cheap ESPs are, maybe just two and you're set up.

pilot1981, 2022-12-08:
Thanks for reply, you know how to connect directly HA and NUKI smart lock 3.0 pro (WIFI version)?

I don’t find API token…

pilot1981, 2022-12-08:
I found a way to use API token via Julie web access.

you think using ESP32 I can save battery of NUKI smart lock 3.0 pro?

technyon, 2022-12-08:
If you disable Wifi and use the ESP instead, you'll save battery.

pilot1981, 2022-12-08:
Hi,
Today I tested your firmware with my NUKI smart lock 3.0: installation works fine and MQTT and NUKI connection are OK, but when I send unlock command from HA: it don’t works!

how I can troubleshoot it?

technyon, 2022-12-08:
You did successfully pair it? Do you values reported in HA like battery level? If yes, browse the MQTT tree and see if the lock/action node is set when you send the unlock command from HA.

pilot1981, 2022-12-08:
[image]
[image]
[image]

pilot1981, 2022-12-08:
State of casetta is wrong:
[image]

technyon, 2022-12-08:
Can you try to directly set the "lock/action" node to "lock" or "unlock" and see if you have delays. The question is where the action gets lost. Doesn't HA send the action, does the MQTT broker somehow not forward it (maybe because MQTT is disconnected at the time), or does the command get sent to the ESP, but somehow bluetooth communication fails. Any of those can fail, and we need to isolate the broken link.

pilot1981, 2022-12-08:
Hi,
I added PIN code in credential section of NUKIhub.

I tried to launch service unlock from Lovelace and from tool services of Ha But nothing happened.

I red mqtt messages from broker, I suppose connection are good.

I attach MQTT messages:

[image]
[image]
[image]
[image]
[image]
[image]

technyon, 2022-12-09:
I think you should take care of the "Error: This client is not authorized."

pilot1981, 2022-12-09:
sorry, this one is a client mis configuration...not our problem...

pilot1981, 2022-12-09:
I also have to setup username e password?

technyon, 2022-12-10:
Hi, if your MQTT broker is configured with a username and a password, you'll have to give it to NUKI Hub so it can connect to the broker. However, it says "MQTT connected: Yes" in your picture, so all is fine. I guess the error message is about some other device.

Can you download some MQTT client and set the "lock/action" node as described. This would really help because it takes home assistant out of the equasion and we can see if this is HA related or not.

pilot1981, 2022-12-10:
Sure, I go to use an MQTT client to send unlock command to MQTT broker Mosquitto Installed on my HA!

I let you know!

pilot1981, 2022-12-10:
Hi,
I tried and all works fine:

[image]

pilot1981, 2022-12-10:
[image]

pilot1981, 2022-12-10:
ok, now all works also from HA. I don't changed anything...I don't understand...

thanks for your help and your work!

closed 2022-12-10 by pilot1981

technyon, 2022-12-10:
ok glad to hear.