technyon/nuki_hub #55: Unable to add new keypad user

DaveGalbraith, 2022-12-14:
I am having trouble getting the add a new keypad user process to work. Everything else seems to work well (great job btw!), for example, I can delete a user that I have previously created using the Nuki app. But If I run a HA script to add a new user I get a failed commandResult (monitored using MQTT Explorer). I am running the Mosquito broker add-on in HA. Below are my script yaml's:

```
alias: "Nuki: Add New User Code"
sequence:
  - service: mqtt.publish
    data:
      topic: nuki/keypad/command/name
      payload: fred
      qos: "2"
  - service: mqtt.publish
    data:
      topic: nuki/keypad/command/code
      payload: "123456"
      qos: "2"
  - service: mqtt.publish
    data:
      topic: nuki/keypad/command/enabled
      payload: "1"
      qos: "2"
  - service: mqtt.publish
    data:
      topic: nuki/keypad/command/action
      payload: add
      qos: "2"
mode: single

```

```
alias: "Nuki: Delete User Code "
sequence:
  - service: mqtt.publish
    data:
      topic: nuki/keypad/command/id
      payload: "4"
      qos: "0"
  - service: mqtt.publish
    data:
      topic: nuki/keypad/command/action
      payload: delete
      qos: "0"
mode: single

```
Any help or pointers to aid debugging will be greatly appreciated.

Also, I did not see any way to set time limited access on the keypad codes. Is this possible using nuki_hub and, if not, are there plans to introduce such functionality?

technyon, 2022-12-14:
I don't see anything obvious that's wrong with the script. I tried manually setting nodes the same way and it works for me. Does it work when you set it manually using MQTT explorer?

I've not added the time limited access functionality because the API is a bit tedious ... it requires six fields to be set:

https://developer.nuki.io/page/nuki-smart-lock-api-2/2#heading--add-keypad-code

But if there's a need I could add it.

DaveGalbraith, 2022-12-14:
Yeah, I started off using MQTT Explorer but that didnt work either. Interestingly I see data appearing on the command node continuously, multiple times per second as below. Is that to be expected?

[image]

My use case depends on being able to set the time limited access functionality, so I definitely have a need (although that's kinda moot if I can't set up new users in the first place)

technyon, 2022-12-15:
No that's not supposed to happen. I'll check the code later. It's supposed to ignore the UnknownCommand string to prevent a neverending loop of receiving and sending UnknownCommand. For now maybe disconnect the ESP, and write a valid command in there, then re-connect the ESP.

DaveGalbraith, 2022-12-15:
Ahh Ok, well at least we are on to something. Your disconnect suggestion didn't work I'm afraid.

technyon, 2022-12-15:
Which node exactly do you get constant traffic on? The "action" node? Whenever you write something to the action node, two things happen:

- The action node itself will be reset to "--"
- The commandResult node gets updated.

Is the UnknownCommand from you testing something? You should only ever write "add", "delete or "update" to the action node, anything else will result in an UnknownCommand. If that happens unintentionally, someone (HA ?) is writing something to the action node. It's just not very obvious, because it gets reset to "--" immediately by the ESP.

The attached firmware outputs what it receives on the action node over the serial line.

[nuki_hub.zip](https://github.com/technyon/nuki_hub/files/10240534/nuki_hub.zip)

DaveGalbraith, 2022-12-15:
Hopefully this video makes it obvious which nodes. If it's coming from HA it's not something I've (intentionally ) done. Is there a way to tell where it's coming from (apologies I am very new to MQTT)?

https://user-images.githubusercontent.com/30226844/207974947-dafee959-75d6-4fe2-94d2-fee4679550dd.mp4

technyon, 2022-12-15:
From what I can see I think all nodes are updated. The most plausible explanation is indeed that something is writing to the action node, because that resets all the other nodes (showing as an update in MQTT explorer).

You can:
- Have a look at the history of the nodes, especially the action node
- write something into the commandResult node (e. g. "###"). If it's immediantely reset to UnknownCommand, you know the ESP is updating the node.

To really tell which client is updating the node I guess you have to check your MQTT broker, so there's general answer. It depends on whatever broker you are using.

DaveGalbraith, 2022-12-15:
> Have a look at the history of the nodes, especially the action node

 The history of these nodes just shows the same thing repeated again and again (as per this screen shot):

[image]

> write something into the commandResult node (e. g. "###"). If it's immediantely reset to UnknownCommand, you know the ESP is updating the node.

It is immediately reset to UnknownCommand when I write to commandResult.

And I am using Mosquito, when I set the logs to debug this is what I see receiving and transmitting:

[image]

technyon, 2022-12-16:
The log is quite normal, the firmware publishes these nodes on a regular interval. Can you check the logs again and pipe it through "grep keypad"? Also what's the history on the "action" node?

technyon, 2022-12-17:
Can you try 6.3 ? I've added a check there to ignore "--" in the action node.

DaveGalbraith, 2022-12-17:
I've uploaded 6.3, the continuous updating of the keypad nodes has stopped, however, if I run my script to add a new keypad user, I can see those values being published but there is no response on the commandResult node (and no new user appears in the nuki app)

technyon, 2022-12-17:
Just to make sure, you did enter the PIN on the ESP?

DaveGalbraith, 2022-12-17:
Hmmm, I'm not sure what you mean. I have previously entered the PIN in the credentials page (and I've now just done it again on your prompt, in case you meant after updating the firmware). In any case, there's still no response on thecommandResult node after I run my add user script.

technyon, 2022-12-17:
I just noticed that I made a really stupid mistake when trying to fix things. Keypad commands didn't work at all with 6.3 and 6.4, it's fixed now in 6.5. I reviewed the code, I have no explanation why nodes would be constantly updated ... it doesn't happen for me. I wonder if this somehow has to do with the broker being used.

DaveGalbraith, 2022-12-17:
I do get a commandResult this time, but I am afraid its "failed"

technyon, 2022-12-17:
ok ... does it always return failed? Failed usually means something went wrong with the bluetooth communicaiton.

DaveGalbraith, 2022-12-17:
Yes, it fails every time I try (I've just checked 10 times or so just in case)

technyon, 2022-12-18:
ok, there are a few reasons why it could fail:

- bluetooth communication failed. Unlikely if everything else works.
- PIN missing or wrong ... but you checked that already
- Parameters wrong. You have to give a name and a 6 digit code. No zeroes allowed in the code, for obvious reasons :)

Other than that I'm out of ideas

DaveGalbraith, 2022-12-18:
> bluetooth communication failed. Unlikely if everything else works.

Yeah, and I've just double checked that by deleting a user

> PIN missing or wrong ... but you checked that already

Well, I've re-entered it, but I saw no way of actually checking it (if I deliberately add in a wrong pin I get no notification of that)

> Parameters wrong. You have to give a name and a 6 digit code. No zeroes allowed in the code, for obvious reasons :)

Yes, indeed.

One point of clarification, I don't get a "failed" commandResult now, I get nothing at all when I try to add a new user.

technyon, 2022-12-18:
Please flash the attached firmware and check the serial logs. If successful it should look like this:

```
# KP action received: add
## KP has keypad: 1
## KP keypad enabled: 1
## KP has keypad: ADD
## KP name: the name
## KP code: 111222
Add keypad code: 1
```

One thing I noticed: It seems the lock rejects certain simple codes. I tried "123456" for testing, and I consistently get "failed" as a result. Trying the same in the NUKI smartphone app, it says "entry codes must not start with 12" ... so whatever code you're trying to add, better check first in the smartphone add if it's a valid code.

[nuki_hub-dbg-1.zip](https://github.com/technyon/nuki_hub/files/10253975/nuki_hub-dbg-1.zip)

technyon, 2022-12-18:
I just noticed I put 123456 as an example in the readme ... I've updated it to something that works.

DaveGalbraith, 2022-12-19:
I changed the code to one that didn't start "12..." and it now works. I cant believe it was that that had me foxed all along. Thanks for your help and persistence with this, it works great now.

Now, if you ever get round to adding in the time limited access ...

knorrre, 2023-01-31:
Wild guess: is it a keypad 2.0?

technyon, 2023-01-31:
Nope, this was related to using the example PIN I provider (123456). The Locks firmware doesn't allow PINs starting with "12" to prevent people from using 123456 as a PIN. I didn't know that at the time of writing of the README, but I've since corrected it, and the issue could be resolved.