technyon/nuki_hub #57: Support for nuki door sensor

KitHubek, 2022-12-17:
is possible to add support for door sensor?

https://nuki.io/en/door-sensor/

technyon, 2022-12-17:
The door sensor is already supported.

KitHubek, 2022-12-17:
i think this is binary_sensor.NAMEi_door_sensor ?
I mean about battery door sensor

technyon, 2022-12-18:
Hi, sorry I misread your post. There's no direct support for the battery level in the bluetooth API, only a bit signalling critical battery state. Yes, support for this can be added, but it'll only be battery good yes/no.

KitHubek, 2022-12-18:
ok, if possible, please add this option. Thi will be battery state: critical or good? If this is possible, this will better than nothing. thanks

technyon, 2022-12-18:
Check Release 6.6

KitHubek, 2022-12-18:
i mean that this sensor, door sensor battery is: binary_sensor.NAME_keypad_battery_low ?
And if yes, what if i install keypad v2 - i planned it

technyon, 2022-12-18:
Yes it should show up under a proper name though. I haven't seen any API changes for v2, so my guess is it'll be the same. If you try it please let me know if it works.

KitHubek, 2022-12-18:
ok, i will write when i got one. thanks

KitHubek, 2023-01-06:
fornow closed

closed 2023-01-06 by KitHubek