technyon/nuki_hub #58: Lock state from opened to closed took 30m min, door sensor state change immediately

KitHubek, 2022-12-17:
Please see screens. Lock entitie change state from unlocked to locked took 30min. Door sensor changed state immediately

[image]
[image]
[image]

technyon, 2022-12-31:
Hi,

I think this is what happened:

- The ESP was notified via a BLE beacon from the lock that the state changed
- As soon as that happens, the ESP initate communication with the lock to query the new state
- The communication failed for some reason
- Half an hour later the interval for polling kicked in and the lock state was successfully queried this time (This is the interval configured by "Query interval lock state")

A fix for this would be to implement an automatic retry if the bluetooth communication fails as mentioned in issue #62.

Does this happen regularly for you?

SergioWarriorPeru, 2023-01-04:
A question, what happen if I put "Query interval lock state (seconds)" in time shorter, for example, 50 sec?...
I have realized that my door sensor, when this Query interval is 1800 sec, some times last 30 min to update state (open or close)....I have put this interval in 50 sec and now it update in worst case in 50 sec...no always but sometimes.
Lock state is immediate

technyon, 2023-01-05:
Well, my theory was correct then. You can lower the interval which, it comes though with lower battery life because the lock of course need to use power to talk bluetooth