technyon/nuki_hub #61: auto discovery fails - seems like a race condition

regevbr, 2022-12-20:
I can't get auto discovery to work for me - I keep getting 
`Unable to setup HASS. Invalid config received.`
But it seems that the config is then received properly (see below)
Can you please help?

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0
Network device: Builtin WiFi
MQTT without TLS.
*wm:[1] AutoConnect 
*wm:[2] ESP32 event handler enabled 
*wm:[2] Setting Hostnames:  nukihub
*wm:[2] Setting WiFi hostname 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to SAVED AP: notforanyone
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] AutoConnect: SUCCESS 
*wm:[2] Connected in 1634 ms
*wm:[1] STA IP Address: 192.168.31.91
WiFi connected: 192.168.31.91
Host name: nukihub
MQTT Broker: 192.168.31.92:1883
NUKI Lock enabled
Lock state interval: 1800 | Battery interval: 1800 | Publish auth data: no
NUKI Opener disabled
Presence detection timeout (ms): 60000
Attempting MQTT connection
MQTT: Connecting with user: mqtt
MQTT connected
Nuki start pairing
Nuki paired
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
Unable to setup HASS. Invalid config received.
lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
Nuki lock state: locked
Reading config. Result: 1
Reading advanced config. Result: 1
```

Mincka, 2022-12-20:
I can also see this in 6.6. Publishing of autodiscovery is not done.
I tried to change the discovery topic but it's was not published. I reverted back to 6.1 and it works fine.

technyon, 2022-12-24:
Please try the attached binary.

[nuki_hub-dbg-ha-setup-1.zip](https://github.com/technyon/nuki_hub/files/10298214/nuki_hub-dbg-ha-setup-1.zip)

regevbr, 2022-12-24:
Can i upgrade and then downgrade without issues via Ota?

technyon, 2022-12-24:
yes no problem. It's not really a downgrade, it's version 6.6 but I changed the setup procedure for HA. It used to request the config twice, which is unnecessarry but probably leads to this race condition.

regevbr, 2022-12-24:
Now it succeds and I see the messages in mqtt thanks!

technyon, 2022-12-24:
Check Release 6.7

closed 2022-12-24 by technyon