technyon/nuki_hub #62: add retry mechanism

regevbr, 2022-12-22:
Oftentimes, lock and unlock commands fail. When using automation this isn't very pleasant :-) to overcome it I created a somewhat "complex" NodeRed flow that handles command retries incase of failures to ensure that the commands sent are executed. 
It will be great if we can support such a feature built-in.

[image]

technyon, 2022-12-23:
Hi, I have to fix some other things before I can look into this. Maybe it's also a good idea to figure out why it fails that often for you, for me it works almost always. Maybe the connection between the lock and the ESP could be improved (move the ESP closer to the lock)?

regevbr, 2022-12-23:
It is very close... How can i get it to output relevant logs?

technyon, 2022-12-23:
Add the moment there's unfortunately no logging over network yet. You can read output from the serial line for example with hterm (set to 115.200 baud)

Mincka, 2022-12-24:
@regevbr Maybe you can start by looking at the MQTT topics with MQTT Explorer. You will see how the topics are updated. There is a `commandResult` value that can be interesting to look at for things like time out issues.
[image]

regevbr, 2022-12-24:
I just see failed not tine out

technyon, 2022-12-24:
@Mincka What happened to your lock that you had a motorPowerFailure?

Mincka, 2022-12-24:
> @Mincka What happened to your lock that you had a motorPowerFailure?

Hardware defect of the lock. It has been replaced by another one by Nuki since I was in the first year of warranty. That's the new one that is not compatible for some reason with the latest version of nimble...

I don't know the exact reason of the hardware failure. In the past, I had a motor jammed issue once because my door was not properly closed but the lock was still working fine after that. A few weeks later, it just stopped working with this specific issue and an error code in the app saying I need to contact the support. Anyway, the Nuki support was excellent and they send me back a new lock (that I guess is a refurbished one, but it works great).

regevbr, 2022-12-24:
> Add the moment there's unfortunately no logging over network yet. You can read output from the serial line for example with hterm (set to 115.200 baud)

I only see "failed". completionStatus is uncahnged

regevbr, 2022-12-24:
@technyon just installed latest version - log topic is not being published to... so I can't find out why I get errors from time to time

technyon, 2022-12-24:
Please go to the MQTT Configuration and enabled logging there.

regevbr, 2022-12-26:
Thanks but nothing special in the logs is showing

technyon, 2022-12-31:
Please test the attached binary. It retries up to three times to send the lock action, and retries to query the lock state every 10 seconds if it fails.

[nuki_hub-retry-1.zip](https://github.com/technyon/nuki_hub/files/10327472/nuki_hub-retry-1.zip)

regevbr, 2023-01-02:
Thanks I just perfomred on OTA I will report back soon

regevbr, 2023-01-04:
It happened to me twice where the retry mechanism kicks in and then the whole thing gets stuck and unresponsive...

The logs I received:
2023-01-04 10:17:01.897 WARNING (MainThread) [nuki_hub] Nuki lock state update: unlocked
2023-01-04 10:17:12.975 WARNING (MainThread) [nuki_hub] Lock action received: lock
2023-01-04 10:17:17.988 WARNING (MainThread) [nuki_hub] Lock action result: failed
2023-01-04 10:17:18.078 WARNING (MainThread) [nuki_hub] Lock action failed, retry 0
2023-01-04 10:19:51.163 WARNING (MainThread) [nuki_hub] Lock action received: lock

As you can see there are no more retyr logs and when I close the lock using the physical nuki button, the hub doesn't get updated that the lock was locked and I need to restart it.

Mincka, 2023-01-11:
Would it be possible to have the retry mechanism implemented for the next release or rebased for 6.9 in this thread for testing?

The 6.9 version is working great most of the time for now.
I would like to confirm that the retry implementation would help for one of my lock that get a `commandResult = failed` `Lock action result: failed` from time to time. It usually works on the second attempt.

Mincka, 2023-01-12:
In my case, the lock is still responsive and one retry attempt in case of `commandResult: failed` is usually enough to succeed. I think it's because of the BT connection that is a little weak. I don't think the status query is needed at first (and we saw that's better to do small changes in each release). I don't know how you implemented it but a new attempt of the same command in case of `failed` status would be great.

technyon, 2023-01-12:
Yes I thinking limiting the number of retries makes sense anyway.

@regevbr When testing this, did you disable your NodeRed flow?

regevbr, 2023-01-12:
> Yes I thinking limiting the number of retries makes sense anyway.
> 
> @regevbr When testing this, did you disable your NodeRed flow?

Yes of course... For me without the fix + node red flow - everything is operating flawlessly. Sadly the fix here caused the crash for some reason. If you can figure out a way why that happened I will be glad to test it out again. In the meanwhile maybe @Mincka  can also give it a go and report back if it gets stuck for him as well or not.

@technyon thanks for looking into this :-)

Mincka, 2023-01-12:
Sure, I will test a retry mechanism if rebased for 6.10 or included in the next version.
I am not sure to get what exactly was added by @technyon in the test version of this thread. It looks like two different things, the retry on failure and the "retries to query". Maybe that's the second part that is not necessary and causing issues in your case. 

Also, I don't know what would be the ideal delay to add between attempts. I hope it's enough to wait 1 second but in the meantime I don't think it should be less than that. Being able to have this ("Delay between retry attempts (in seconds)" as a parameter for testing would be the icing on the cake.

technyon, 2023-01-14:
I gave it another try. You can configure number of retries and delay, please let me know how it works.

[nuki_hub-6.10-retry-1.zip](https://github.com/technyon/nuki_hub/files/10417010/nuki_hub-6.10-retry-1.zip)

Mincka, 2023-01-14:
Thank you so much for this. :)

I used this to begin:
[image]

It worked on the first try! Then, I tried again and this time it seemed that it did not even try to retry. Then again, and it worked. Even if the last command result was `failed`, when it was published again, it worked.

I continue to test. So for now, it worked for 2 failed unlocking actions but failed for the locking one.

Mincka, 2023-01-14:
I can confirm that from times to times, the retry mechanism is not triggered but I am not sure why.
I updated the number of attempts to 3 and I had one case where all 3 retry attempts failed. I'm going to increase up to 10, still keeping 100 ms delay.

Is it possible to add the retry count in the MQTT message like you did before? It's useful to get the state updated in HA and get a new event log.

[image]

Something like:
`Last command failed, retrying (2/3) after 100 milliseconds. Lock action result: failed`

Mincka, 2023-01-14:
Ouch, I just tried to unlock one more time and it made the ESP reboot and spam the 10 retries.
[image]
[image]

technyon, 2023-01-14:
This one should work :). Also added a retry counter MQTT node.

[nuki_hub-6.10-retry-2.zip](https://github.com/technyon/nuki_hub/files/10417346/nuki_hub-6.10-retry-2.zip)

Mincka, 2023-01-14:
Thank you. ☕;-)
First attempt, 2 retries because of `commandResult = failed` then `commandResult = error` and in this case, it pushed `failed` in `retry`.
[image]

Mincka, 2023-01-14:
Despite this one glitch, all other retry mechanism triggers managed to succeed in the end with 1 to 5 retries needed.
I had a look at MQTT history and when `commandResult = error` after the third or fourth attempt, it does not wait between new attempts and looped to 10 in 600 ms I guess (100ms x 6), so the retry attempts are exceeded and the code push `retry = failed`. That's the same kind of behavior I had on reboot previously. Maybe you need to handle differently `failed`, that needs retries vs other error codes. You're almost there!

technyon, 2023-01-14:
Yes you're right, the delay didn't work, I fixed it.

[nuki_hub-6.10-retry-3.zip](https://github.com/technyon/nuki_hub/files/10418351/nuki_hub-6.10-retry-3.zip)

Mincka, 2023-01-14:
100% success on 10 locks / 10 unlocks both locks with different kinds of error (error, timeOut, failed).
The retry mechanism even helped to update a status that was not correct after the lock was triggered.
I need to confirm this other a few days but this feature helps so much for the reliability of the actions. 👍🤩

technyon, 2023-01-14:
Released as 6.11

closed 2023-01-14 by technyon

Mincka, 2023-01-25:
Just for reference, it seems a similar issue was discussed upstream in the `NukiBleEsp32` project:
https://github.com/I-Connect/NukiBleEsp32/issues/24#issuecomment-1265349390

Maybe that's the same thing, but it's now it handle in this lib. I can confirm again that it helps a solve for reliability.