technyon/nuki_hub #64: ESP offline

pilot1981, 2022-12-26:
Hi,
I added restart option when WIFI disconnect, but ESP continue to remain disconnected from WIFI network.

how I can solve it?

pilot1981, 2022-12-26:
How do you suggest to setup value of restart if Bluetooth beacons not received?

technyon, 2022-12-28:
Hi,

if nothing else works you can set the "restart timer" so the esp reboots after a defined time. Also upgrading to 6.7 could make sense, since I've moved to the the ESP core, there's a remote chance they've fixed something the Wifi stack.

Another option is to buy a W5500 module and connect the ESP via cable to the network.

You only have to configure the restart for missing bluetooth beacons if you have problems with bluetooth functionality. If it's your Wifi that disconnects, this option won't help you.

pilot1981, 2022-12-29:
I already activated option of automatic restart when WIFI disconnect…but it don’t solved my problem.

technyon, 2022-12-30:
There's another option called "restart timer", which restarts the ESP at defined intervals.

pilot1981, 2022-12-30:
Thanks.

which difference there is between “network timeout untile restart” and “restart timer” ?

technyon, 2022-12-30:
network timeout until restart: after x seconds of the mqtt server not being reachable, the ESP reboots
restart timer: The ESP reboots x minutes after booting. If you set to to 60, it reboots once every hour

technyon, 2023-01-10:
Hi, could you figure out a solution?

pilot1981, 2023-01-10:
To solve my issue of WiFi disconnection, I solved using timer feature…

thanks

closed 2023-01-10 by pilot1981

technyon, 2023-01-10:
ok, glad to hear