technyon/nuki_hub #67: Add a way to configure "continuous mode" ("Dauermodus") for Opener

mundschenk-at, 2023-01-01:
Currently there appears to be no way to enable or disable "continuous mode" ("Dauermodus") for the Opener. Also when "continuous mode" is enabled, the "lock" created for the Opener is still shown in a "locked" state.

technyon, 2023-01-02:
Hi,

You can enable or disable CM (continuous mode) by sending the "activatCM" and "deactivateCM" commands. You're right this isn't yet reflected in the state, because of the strange way the bluetooth API signals this. Instead of changing the state it sets some bit somewhere else to show that CM is active - this could be easily mapped though by the ESP to reflect it in the state.

mundschenk-at, 2023-01-02:
Ah, thanks I must have overread that in the documentation. Too me, ideally there would be a separate state topic for CM itself (which should be mapped to HA autodiscovery as a configuration switch) and it should be reflected in the `binaryState` of the lock to accurately reflect whether the controlled door is "(ring to) open".

technyon, 2023-01-02:
I think it still has to be handled via commands, because it interacts with the other commands ... for example writing "activateRTO" should disable CM if active.

If it appears as a seperate topic, it would imply this should work:

- activate RTO
- activate CM
- deactivate CM
- device switches back to RTO because it was active before

What would actually happen though it would switch back to "locked", which isn't very intuititive.

mundschenk-at, 2023-01-02:
Are you sure this is the way it works? I set CM via the official app and then activated and deactivated RTO via unlocking the lock entity in HA -  CM was still active when I checked in the app.

technyon, 2023-01-02:
Not a 100% :D. I'll do some reverse engineering to see how the firmware really behaves - it's not clearly stated in the documentation.

technyon, 2023-01-10:
With the latest release it's not at least shown in the state node, and the binary state is updated accordingly. I'll have to see about adding a sensor, the API is quite confusing about this. It seems the NUKI App can indeed update CM and RTO independently.

mundschenk-at, 2023-01-29:
Would it maybe be better to map the behavior of the HA `lock` entity to enabling/disabling CM instead of RTO (which has a maximum duration of 1 hour) for the `lock`/`unlock` services?

PR #296 (2024-02-18): Enable Additional Lock states in Home Assistant + Make Opener Continuous mode an option for lock/unlock
**Allow Home Assistant to show intermediate states (unlocking/locking/opening) and jammed state**

Note: Not all intermediate states are captured by the Hub and these will not show up in Home Assistant. This change will just make Home Assistant more accurately reflect the state of nuki/lock/state instead of only being able to show locked/unlocked.

Closes #253, Closes #163

**Add a separate MQTT topic for continuous mode state, distinct from lock state
Make Opener Continuous mode (instead of [... cut]
closed 2024-02-18 by technyon