technyon/nuki_hub #68: Presence detection enabled even when disabled in config

ruimarinho, 2023-01-02:
Hi @technyon,

I have disabled `Presence detection` by setting it to -1.

[image]

However, I still see messages getting published to MQTT under `nuki/presence/devices`.

technyon, 2023-01-02:
It just rechecked it, it works as intended for me. Are you sure those are new messages? Maybe just try to change the value again, basically anything smaller than zero should disable the presence detection.

SergioWarriorPeru, 2023-01-04:
A question, What is Presence detection timeout?

technyon, 2023-01-04:
The presence is detected by a receiving bluetooth beacons. If no beacons are received for a certain time, the device is assumed to be absent. The presence detection time is exactly this "certain time".

technyon, 2023-01-10:
Hi, where you able to disable it in the meantime?

ruimarinho, 2023-01-10:
I've just set it to -2 and I stopped seeing it @technyon. Setting it back to -1 fixed it for good. Also tried enabling it again and then setting it to -1 and it also worked. I'm pretty sure the bug is still there but right now I'm unable to replicate it. 

Does having it enabled contribute to a shorter battery time or is this BLE scan happening in the background anyway?

closed 2023-01-10 by ruimarinho

technyon, 2023-01-10:
It won't have any effect on the battery life, as it's the ESP doing it and not the lock. And yes it has to scan all the time anyway, because it's the mechanism how the lock signals that there's an update. If you don't need, you sure can disable publishing it, it'll at least save some network traffic.

Must've been some kind of fluke why it didn't get disabled, can't really say why.