technyon/nuki_hub #70: Nuki 3.0 stopped working since 6.2+

lucaoliano, 2023-01-04:
Hi @technyon,
first of all thank you for your great work.

I have a Nuki Lock 3.0 and it doesn't work anymore since 6.2+

This morning I tried unsuccessfully the last version 6.8. The lock/unlock works only once then I have "Lock action result: error". For deeper investigation I've recompiled the firmware with debug log:

```

16:04:42.398 > Lock action received: unlock

16:04:42.412 > [ 40633][E][NukiBle.hpp:8] executeAction(): Lock Heartbeat timeout, command failed
16:04:42.420 > Lock action result: error

```

I tried downgrading the libraries NimBLE-Arduino and nuki_ble and I had success only with NimBLE-Arduino 1.4.0 and nuki_ble d63a9ff, same versions of 6.1.

Here the logs of my tests:

- NimBLE-Arduino 1.4.0 and nuki_ble be1fa97
[image]

- NimBLE-Arduino 1.4.1 and nuki_ble d63a9ff
[image]

adizarko123, 2023-01-04:
This also happened to me with Nuki 3.0
Downgrading to 6.1 solved the issue.

technyon, 2023-01-04:
Did you try this binary? My guess is it's the changes in the state machines, especially since you get timeout errors.

[nuki_hub-6.7-rvt-cmd-stm.zip](https://github.com/technyon/nuki_hub/files/10326286/nuki_hub-6.7-rvt-cmd-stm.zip)

Christoph22, 2023-01-05:
> Yes, I had already tried that version without success. Here the logs
> 
> ```
> MQTT connected
> Nuki start pairing
> Nuki paired
> lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
> Reading config. Result: success
> Reading advanced config. Result: success
> Lock action received: lock
> Lock action result: error
> ```
I have the same problem mit my nuki 3.0
Tested this binary too, also no success

thx for your great work

technyon, 2023-01-05:
I honestly have no idea anymore, this all seems completely random. And I can't reproduce it, fixing it will be difficult like this.

For now I've
- reverted NimBLE back to 1.4.0
- Created a branch from the NUKI library from the last known good commit and applied the needed changes to make it compile again, which are minimal.

Can you test the attached version? If it work, I'd release it as 6.9 so there's no broken binary out as an official release.

[nuki_hub-6.8-rvt-all.zip](https://github.com/technyon/nuki_hub/files/10354978/nuki_hub-6.8-rvt-all.zip)

Christoph22, 2023-01-05:
Thx, this version works for me !
But i only tested lock / unlock...

technyon, 2023-01-05:
Are you running the lock on the latest firmware? There was an update not too long ago.

Christoph22, 2023-01-06:
Yes, firmware is 3.4.10, the Lock is only a few weeks old.

KitHubek, 2023-01-06:
iam using lock 3.0 with  6.6 and confirm that WORKIN ok, so problem in this thread isnt global for sure. HA ver core-2023.1.0 instalation tyope: HA OS

lucaoliano, 2023-01-06:
I'm on the latest Nuki fw version 3.4.10, I just tried the version 6.8-rvt-all and it doesn't work for me, below the logs:

```
3:03:09.517 > MQTT connected
13:03:09.517 > Nuki start pairing
13:03:09.517 > Nuki paired
13:03:30.742 > Reading config. Result: Lock action received: lock
13:03:49.223 > timeOut
13:03:49.225 > Reading advanced config. Result: timeOut
13:04:20.460 > Lock action result: timeOut
```

I think that is something related the hw. I did another test, I have flashed the "official" 6.8, rebuilt for ESP32 solo 1, on a Shelly Plus 1 and works well, instead on two AZ-Delivery ESP-32 Dev Kit C V4 doesn't work.

technyon, 2023-01-07:
It seems to be it's multiple issues that occur for some and not for others ... not so sure what to do now.

@Christoph22 Could you give another binary a try. This one has NimBLE 1.4, but only parts of the nuki library reverted.

[nuki_hub-rvt-refact-busy.zip](https://github.com/technyon/nuki_hub/files/10365987/nuki_hub-rvt-refact-busy.zip)

lucaoliano, 2023-01-07:
I tried the binary `nuki_hub-rvt-refact-busy`, it has worked just twice then error again.

```
13:00:43.946 > MQTT connected
13:00:48.910 > Nuki start pairing
13:00:54.137 > Nuki start pairing
13:00:56.354 > Nuki paired
13:00:56.408 > Nuki lock state update: unlocked
13:00:56.614 > Reading config. Result: success
13:00:56.774 > Reading advanced config. Result: success
13:01:05.689 > Lock action received: lock
13:01:06.959 > Lock action result: success
13:01:17.984 > Lock action received: unlock
13:01:18.729 > Nuki lock state update: locked
13:01:18.903 > Lock action result: success
13:01:30.130 > Nuki lock state update: unlocked
13:01:32.899 > Lock action received: lock
13:01:32.921 > Lock action result: error
13:01:37.741 > Lock action received: unlatch
13:01:37.751 > Lock action result: error
```

Christoph22, 2023-01-07:
> It seems to be it's multiple issues that occur for some and not for others ... not so sure what to do now.
>
> @Christoph22 Could you give another binary a try. This one has NimBLE 1.4, but only parts [...]

@technyon binary works for me, updated over OTA
the interface shows the version 6.8-rvt-all, is that ok ?

technyon, 2023-01-07:
no in that case OTA didn't work ... just try again. Should show like in zip filename.

Christoph22, 2023-01-07:
ok got it, works.

> 16:32:27.750 -> MQTT connected
16:32:32.673 -> Nuki start pairing
16:32:32.673 -> Nuki paired
16:32:33.929 -> Nuki lock state update: unlocked
16:32:34.139 -> Reading config. Result: success
16:32:34.241 -> Reading advanced config. Result: success
16:32:34.275 -> HASS setup for lock completed.
16:32:42.968 -> Lock action received: lock
16:32:44.296 -> Lock action result: success
16:32:50.722 -> lld_pdu_get_tx_flush_nb HCI packet count mismatch (1, 2)
16:32:51.805 -> Nuki lock state update: locked
16:32:56.058 -> Lock action received: unlock
16:32:57.936 -> Lock action result: success
16:33:04.849 -> Nuki lock state update: unlocked`

technyon, 2023-01-07:
OK, finally we're getting somewhere. Could you try one other binary? I'd like to know if nimble 1.4.1 makes any problems. This binary is the same as before except for the nimble is upgraded to 1.4.1.

[nuki_hub-6.8-rvt-refact-busy-nimble-1.4.1.zip](https://github.com/technyon/nuki_hub/files/10367011/nuki_hub-6.8-rvt-refact-busy-nimble-1.4.zip)

Christoph22, 2023-01-07:
works also well, no error

technyon, 2023-01-07:
ok, thanks. I'll get in touch with the nuki lib maintainers.

lucaoliano, 2023-01-08:
I also tried last binary without success. Like the previous version lock/unlock has worked only two times, after I received error for any command.

technyon, 2023-01-08:
But you said your other device works fine right?

lucaoliano, 2023-01-08:
Yes, I tried on Shelly Plus 1, which is the only other ESP32 device I own. The problem with Shelly is that it comes with a unicore version of ESP32, so any new fw version needs to be rebuilt to support that chip. My test was aimed at understanding if the problem is on the Nuki side or on the ESP32 board side. I think the best thing is to find what in the combination between NimBLE-Arduino 1.4.1 and the new version of nuki_ble is causing the timeout I observed. This way we can have a solution that can work on every board so that everyone can benefit from this great project.

What do you think?

technyon, 2023-01-08:
Just checking. It's quite difficult for me to fix it without being able to reproduce it. Then return code "Error" basically means no ble beacons were received from the lock for 30 seconds. If you can compile yourself, you can add a logger where the heartbeat (= beacon received) is update (NukiBle.cpp, "lastHeartbeat = millis();") and see if there's anything coming through.

I have a feeling no heartbeats at all are received, and you manage to open/close exactly twice within those 30 seconds.

lucaoliano, 2023-01-08:
Thank you for your suggestion. I inspected the `NukiBle::onResult` method and what I found is that `isKeyTurnerUUID` is set to `false` so the following `if (isKeyTurnerUUID)` and `lastHeartbeat = millis()` are not activated.

Any idea why this happens?

technyon, 2023-01-08:
Now we're on to something. It tries to match the service IDs, let't figure out what't going wrong there. The keyturner service ID for the lock should be "a92ee200-5501-11e4-916c-0800200c9a66". Let's log both both IDs ... after "std::string serviceUUID = deviceServiceUUID.toString();" put:

      char uuid[100];
      memset(uuid, 0, sizeof(uuid));
      memcpy(uuid, &pHex[8], 32);
      Serial.print(serviceUUID.c_str());
      Serial.print(" | ");
      Serial.println(uuid);

My output looks like this:

a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66

You can ignore the dashes, the loop below should ignore them when comparing the IDs.

lucaoliano, 2023-01-08:
This is my output:

a92ee200-5501-11e4-916c-0800200c9a66 | 01601cf5f0d430060001000102ed9b91

technyon, 2023-01-08:
That's a clear case of WTF. Is it consistently this number, or does it change after reboot (or better disconnect and reconnect from power)? If it changes after reboot it's probably random garbage in memory.

lucaoliano, 2023-01-08:
It is still the same even after disconnecting and reconnecting the power.

technyon, 2023-01-08:
ok thanks

lucaoliano, 2023-01-09:
I added that log in the [nuki_ble @ d63a9ff](https://github.com/technyon/nuki_ble/tree/d63a9ff39b89ffd92431c93c0df90f98d090264c), the one that works for me. The output is still `a92ee200-5501-11e4-916c-0800200c9a66 | 01601cf5f0d430060001000102ed9b91` and obviously  `isKeyTurnerUUID` is `false`, but I can lock/unlock the Nuki without problem.

technyon, 2023-01-09:
This is very strange to say at least. Can you flash the attached binary, I've extended the logging by the bluetooth address. I'd like to make sure that whatever is output there is really coming from the lock. So after flashing could you verify the address matches the address of your lock? You can check the address either in the NUKI App or in a beacon scanner app (nRF Connect for example).

[nuki_hub-log-on-result-1.zip](https://github.com/technyon/nuki_hub/files/10374910/nuki_hub-log-on-result-1.zip)

lucaoliano, 2023-01-09:
The MAC address in the log matches with the Nuki one, but something very strange has happened (at least to my knowledge). After playing around with nRF Connect I started seeing the correct service UUID and the beacon in the log, even though the previous one is still there.

Below the logs:
```
19:08:02.532 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | 01601cf5f0d430060001000102ed9b91
19:08:04.536 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | 01601cf5f0d430060001000102ed9b91
19:08:06.958 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
19:08:06.967 > [123950][D][NukiBle.cpp:225] onResult(): Nuki Advertising: Name: Nuki_33C335C5, Address: 54:d2:72:c3:35:c5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6633c335c5c8
19:08:06.984 > [123962][D][NukiBle.cpp:237] onResult(): iBeacon ID: 004C Major: 13251 Minor: 13765 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
```

technyon, 2023-01-09:
Does the ESP keep receiving the correct or the incorrect uuid? Would be interesting to see what happened after "19:08:06.984". Also, can you somehow correlate receving the correct uuid and the lock actions working or not?

lucaoliano, 2023-01-09:
The ESP keeps receiving both the correct or the incorrect uuid, both are very frequent and the lock actions work correctly not just two times. I think the ESP started getting the correct uuid after I tried to get the Nuki's MAC address from nRF Connect.

These are the actions I did:
1. I couldn't find Nuki's MAC address on nRF Connect, so I connected to device and I clicked everywhere, then I realized that the iOS version of nRF Connect doesn't show MAC addresses
2. I loaded a "BLE Scanner" sketch to the ESP to find the address
3. I loaded the nuki_hub fw and at this point I started seeing the correct uuid

do you see a rationale in this?

technyon, 2023-01-09:
The only thing that could've changed something is whatever you did in the app ... the BLE Scanner is completely passive and doesn't send anything to the lock. No all of this doesn't really make sense.

lucaoliano, 2023-01-09:
I completely agree with you, it doesn't really make sense.

I-Connect, 2023-01-10:
Hi lucaoliano,

Do you have any other devices connecting to the Nuki smart lock, maybe another esp or a Nuki bridge?

And a wild guess... there is no device or application (like nrf connect) spoofing the Mac address of the lock?

We could rule out any influence by other devices by testing the Nuki lock and the esp32 with the nuki_hub fw on a different location (say at least 20 m away from where you currently are testing, including your phone and laptop, or at least with bluetooth disabled)

lucaoliano, 2023-01-10:
Hi I-Connect,
I have a main esp with nuki_hub 6.1 (that was the last working version for me) in a junction box in the wall registered as bridge. Since it's a bit inaccessible, I have another esp for testing registered as app. For each test I turned off the main esp via a circuit breaker. There are no devices that can spoof the Nuki's MAC address.

I repeated the test under these conditions:
- main esp powered off
- test esp with nuki_hub fw 6.8 both registered as bridge and as app
- bluetooth of the laptop switched off
- bluetooth of the phone switched off
- different location

I'm still getting the correct and incorrect service uuid, below one minute of log:
```
13:28:59.506 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | 01601cf5f0d430060001000102ed9b91
13:29:00.530 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
13:29:00.540 > [786648][D][NukiBle.cpp:225] onResult(): Nuki Advertising: Name: Nuki_33C335C5, Address: 54:d2:72:c3:35:c5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6633c335c5c8
13:29:00.556 > [786661][D][NukiBle.cpp:237] onResult(): iBeacon ID: 004C Major: 13251 Minor: 13765 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
13:29:00.567 > 
13:29:01.554 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
13:29:01.563 > [787672][D][NukiBle.cpp:225] onResult(): Nuki Advertising: Name: Nuki_33C335C5, Address: 54:d2:72:c3:35:c5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6633c335c5c8
13:29:01.579 > [787685][D][NukiBle.cpp:237] onResult(): iBeacon ID: 004C Major: 13251 Minor: 13765 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
13:29:01.591 > 
13:29:09.564 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
13:29:09.574 > [795684][D][NukiBle.cpp:225] onResult(): Nuki Advertising: Name: Nuki_33C335C5, Address: 54:d2:72:c3:35:c5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6633c335c5c8
13:29:09.590 > [795696][D][NukiBle.cpp:237] onResult(): iBeacon ID: 004C Major: 13251 Minor: 13765 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
13:29:09.602 > 
13:29:10.589 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
13:29:10.599 > [796708][D][NukiBle.cpp:225] onResult(): Nuki Advertising: Name: Nuki_33C335C5, Address: 54:d2:72:c3:35:c5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6633c335c5c8
13:29:10.614 > [796720][D][NukiBle.cpp:237] onResult(): iBeacon ID: 004C Major: 13251 Minor: 13765 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
13:29:10.627 > 
13:29:11.522 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | 01601cf5f0d430060001000102ed9b91
13:29:12.547 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
13:29:12.557 > [798666][D][NukiBle.cpp:225] onResult(): Nuki Advertising: Name: Nuki_33C335C5, Address: 54:d2:72:c3:35:c5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6633c335c5c8
13:29:12.573 > [798679][D][NukiBle.cpp:237] onResult(): iBeacon ID: 004C Major: 13251 Minor: 13765 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
13:29:12.586 > 
13:29:14.602 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
13:29:14.611 > [800720][D][NukiBle.cpp:225] onResult(): Nuki Advertising: Name: Nuki_33C335C5, Address: 54:d2:72:c3:35:c5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6633c335c5c8
13:29:14.627 > [800733][D][NukiBle.cpp:237] onResult(): iBeacon ID: 004C Major: 13251 Minor: 13765 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
13:29:14.639 > 
13:29:17.574 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
13:29:17.584 > [803692][D][NukiBle.cpp:225] onResult(): Nuki Advertising: Name: Nuki_33C335C5, Address: 54:d2:72:c3:35:c5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6633c335c5c8
13:29:17.599 > [803705][D][NukiBle.cpp:237] onResult(): iBeacon ID: 004C Major: 13251 Minor: 13765 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
13:29:17.611 > 
13:29:19.534 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | 01601cf5f0d430060001000102ed9b91
13:29:21.586 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
13:29:21.596 > [807706][D][NukiBle.cpp:225] onResult(): Nuki Advertising: Name: Nuki_33C335C5, Address: 54:d2:72:c3:35:c5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6633c335c5c8
13:29:21.612 > [807718][D][NukiBle.cpp:237] onResult(): iBeacon ID: 004C Major: 13251 Minor: 13765 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
13:29:21.624 > 
13:29:23.537 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | 01601cf5f0d430060001000102ed9b91
13:29:26.611 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
13:29:26.621 > [812731][D][NukiBle.cpp:225] onResult(): Nuki Advertising: Name: Nuki_33C335C5, Address: 54:d2:72:c3:35:c5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6633c335c5c8
13:29:26.637 > [812743][D][NukiBle.cpp:237] onResult(): iBeacon ID: 004C Major: 13251 Minor: 13765 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
13:29:26.649 > 
13:29:27.543 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | 01601cf5f0d430060001000102ed9b91
[... cut ...]
13:29:59.587 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | 01601cf5f0d430060001000102ed9b91
13:30:00.610 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
13:30:00.620 > [846731][D][NukiBle.cpp:225] onResult(): Nuki Advertising: Name: Nuki_33C335C5, Address: 54:d2:72:c3:35:c5, manufacturer data: 4c000215a92ee200550111e4916c0800200c9a6633c335c5c8
13:30:00.636 > [846744][D][NukiBle.cpp:237] onResult(): iBeacon ID: 004C Major: 13251 Minor: 13765 UUID: a92ee200-5501-11e4-916c-0800200c9a66 Power: -56
```

I-Connect, 2023-01-10:
I will raise a question on the Nuki developer forum. Maybe they can shed some light on this.

Which FW version are you running on the Nuki v3 Smart lock? You can see this in the Nuki app under settings/manage smart lock

Jeroen

lucaoliano, 2023-01-10:
I'm on the latest fw version 3.4.10

I-Connect, 2023-01-10:
https://developer.nuki.io/t/uuid-is-shown-by-nuki-smart-lock-that-we-do-not-understand/19134

lucaoliano, 2023-01-10:
Thanks, I will follow this thread

I-Connect, 2023-01-12:
pls check the nuki developer topic, do you have a Apple Homekit? (and enabled in the Nuki smart lock?)

lucaoliano, 2023-01-12:
I have HomeKit enabled in the Nuki app even though I don't have a HomePod, iPad or Apple TV as a bridge. I also have Home Assistant with HomeKit integration enabled. I have no other integrations.

technyon, 2023-01-12:
The interesting question now: If you disable Homekit, will you stop receiving the "01601cf5f0d430060001000102ed9b91" beacons?

lucaoliano, 2023-01-12:
Tomorrow morning I will test with HomeKit disabled

lucaoliano, 2023-01-13:
After deactivating HomeKit in the Nuki app, the beacon disappears. For completeness the beacon reappears when I reactivate HomeKit and is still the same.

I-Connect, 2023-01-13:
I think the Nuki developer posted our solution, I will check and update if possible.

Glad we are getting somewhere with this and will be a nice fix for a user specific setting :-)

technyon, 2023-01-13:
Do we have to fix anything? As far as I understand both uuid are sent, so we can just ignore the one for home kit?

I-Connect, 2023-01-13:
hmmm, good question, I thought lucaoliano did have an issue until this 
https://github.com/technyon/nuki_hub/issues/70#issuecomment-1376291744

isibizi, 2023-01-14:
I have also the same problem. I can unlock/lock twice and than it stop working for me. I have a smart lock 3 without bridge and homekit is disabled in the app

technyon, 2023-01-14:
If you have homekit enabled, please try disabling it

isibizi, 2023-01-14:
I have do that allready, that not solve my problem.

lucaoliano, 2023-01-15:
That's probably what happened to me, I wasn't receiving the key turner service beacon. Can you install this [binary](https://github.com/technyon/nuki_hub/issues/70#issuecomment-1375887209) and see in the logs which beacons you are receiving? You should see a log line like this one below:
```
13:29:52.601 > 54:d2:72:c3:35:c5 | a92ee200-5501-11e4-916c-0800200c9a66 | a92ee200550111e4916c0800200c9a66
```

phillcz, 2023-01-29:
Hi, I have the same issue with a wemos mini32 and nuki v1 lock. Version 6.1 works fine, versions 6.2, 6.7, 6.10, 8.0 fail.

ESP is a single BLE device connecting to the lock. Nuki runs the latest version 1.14.1. 

I tried factory resetting nuki, no difference. I installed a fresh v8.0, was able to initially pair the lock, but configuration could not be downloaded. The communication with nuki somehow fails.
```
Nuki start pairing
Nuki paired
Reading config. Result: notPaired
Reading advanced config. Result: notPaired
```
Then, right after downgrading to 6.1 it paired, connected and works correctly.

Is there anything I can capture or test?

technyon, 2023-01-30:
Do you have another ESP to try?

phillcz, 2023-02-01:
I powered my current ESP off and tested two other boards. Erased, installed 8.1 and the same thing happened. It paired, but did not communicate.
[image]
[image]

I then powered on my old ESP and it paired, but failed to work. Configuration download failed with error 3. I had to erase the chip, flash it with 6.1, setup and pair again and it started working.

technyon, 2023-02-01:
Can you connect to the broker? The ESP will spend time in the reconnect part for MQTT, which could interrupt bluetooth communication.

technyon, 2023-02-01:
Flash this binary please and post the logs.

[nuki_hub-8.1-dbg-1.zip](https://github.com/technyon/nuki_hub/files/10561885/nuki_hub-8.1-dbg-1.zip)

phillcz, 2023-02-02:
@technyon Here it is: https://gist.github.com/phillcz/993f934d0cbd85d2aa20bff81ac1a4d7
Erased, flashed, connected wifi, configured mqtt, pushed nuki pair button and waited. Some time later I reset the esp using the hw button.

technyon, 2023-02-02:
ok, so the ESP doesn't receive pairing service beacons (No nuki in pairing mode found).

You can try release 8.2. Today I received a new Opener since I had to replace my old one. Pairing was impossible initially, I had to downgrade the NimBLE to make it work again. My logs looked different though, it did receive the pairing service beacons, but couldn't complete the pairing process after that.

phillcz, 2023-02-02:
See line 399 that's when I pressed the button on nuki and it started to pair.

[22 more replies]

closed 2023-04-23 by technyon