technyon/nuki_hub #71: Add to README.md description of all hub options and settings

KitHubek, 2023-01-06:
I see that many people askind here:

```
what do funnction xx?
what is xx?
```

I mean about options in web interface that the user can set or change, for examplpe: 

```
Network Timeout until restart
Restart on disconnect
Restart timer
...
...
...
```

Many options are clear, but i think not for everyone. For me too some options are unclear and incomprehensible and i must do test to know how its working. an explanation of all these settings does not exist in the README file

I think taht good idea is add to README explanation of all options or make some wiki?

bcutter, 2023-02-07:
Exactly. Some features (config options) should really be explained a bit. Either by adding a small subtitle directly to the option or at least at https://github.com/technyon/nuki_hub#readme.

https://community.home-assistant.io/t/nuki-hub-firmware-for-the-esp32/417572/224

technyon, 2023-02-07:
It's on my todo list, but there are also issues to fix, and I have to choose what to do first.

PR #319 (2024-03-07): Readme update
Update for the README and some minor spelling/typo/order changes related to this.

Probably best reviewed at https://github.com/iranl/nuki_hub/tree/readme-update

Closes #71
closed 2024-03-07 by technyon