technyon/nuki_hub #72: Wrong unit of measurement for Signal strength in Home Assistant

alexdelprete, 2023-01-11:
HA is logging an error for the BT/WiFi signal strength sensors:

```
Entity sensor.portoncino_bluetooth_signal_strength (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'db' which is not a valid unit for the device class ('signal_strength') it is using; Please update your configuration if your entity is manually configured
```

The unit of measurement for device class `signal_strength` should be (case sensitive) `dB` or `dBm`

[image]

Documentation: https://developers.home-assistant.io/docs/core/entity/sensor

technyon, 2023-01-12:
Changed and will be included in the next release

closed 2023-01-12 by technyon

alexdelprete, 2023-01-14:
@technyon I upgraded to 6.11, but I still see dB as unit of measurement for both wifi and bt signal strength.

[image]
[image]

Mincka, 2023-01-25:
Actually, `dB` was maybe not the correct unit. I saw this when using ESPHome sensors for Wi-Fi and BT.
They use `dBm` for `SIGNAL_STRENGTH_DECIBELS_MILLIWATT`, for Wi-Fi and BT.
That's also what the devs use in the Android App for Wi-Fi RSSI.

See:
https://github.com/home-assistant/core/blob/master/homeassistant/const.py#L917-L919

alexdelprete, 2023-01-25:
It's a known controversial topic, particularly since WiFi came out:

[image]

To sum it up:

- RSSI should be a positive index: 0-255 (802.11 standard)
- Received Signal Power: is an absolute value, measured in dB, or dBm on a logarithmic scale

So it doesn't matter how HA devs are using it, but what the ESP32 values represent and if it's a log. scale.

technyon, 2023-01-25:
So should we remove the unit altogether?

alexdelprete, 2023-01-25:
> So should we remove the unit altogether?

No, it MUST have a UoM. We should check ESP32/Arduino docs to see what the value represents in this specific case. The unit of measure can be dB or dBm.

Look here, they describe RSSI using dB, then give examples in dBm. :)

[image]

So it really depends on the ESP32 SDK you're using...

alexdelprete, 2023-01-25:
> So should we remove the unit altogether?

nope, we should check what the ESP32 value really represents, and also if it's in dB or dBm. :)

I checked espressif API [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv427esp_wifi_set_rssi_threshold7int32_t) for ESP32:

[image]

So I'd say let's go with `dBm`. :)

alexdelprete, 2023-01-25:
Jan, is this used for automatic reconnects?

[image]