technyon/nuki_hub #74: Please change the device class @technyon

isibizi, 2023-01-13:
Hello, 

in my log i have following error

`Logger: homeassistant.components.sensor
Source: components/sensor/__init__.py:980
Integration: Sensor (documentation, issues)
First occurred: 20:25:20 (2 occurrences)
Last logged: 20:25:20

Entity sensor.wohnungstur_wifi_signal_strength (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'db' which is not a valid unit for the device class ('signal_strength') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.wohnungstur_bluetooth_signal_strength (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'db' which is not a valid unit for the device class ('signal_strength') it is using; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22`

I had this same problem with another sensor. You (@technyon ) must change the device class for this sensor please.

THX!

isibizi, 2023-01-13:
Ups, bug is a known bug :)

https://github.com/technyon/nuki_hub/issues/72

Please close this issue

closed 2023-01-13 by technyon

alexdelprete, 2023-01-14:
> Ups, is a known bug

the device_class is correct, the problem is the unit_of_measurement.