technyon/nuki_hub #77: Loosing pairing after update to 6.11

isibizi, 2023-01-15:
Hello,

i have updated my nukihub to the newest version 6.11 in hope that the error was discribed here https://github.com/technyon/nuki_hub/issues/70 is solved. I have found a new poblem, the web ui show sometimes that the lock is not paired. When i try unlock my lock, it works, that means, the status in the web ui has a bug.

[image]

technyon, 2023-01-15:
Could it be that you opened the web portal right after flashing / turning on the ESP? It needs a few seconds to initialize, after that the lock state should be shown. Or querying the lock state work for some reason you could just wait or try to restart the ESP.

technyon, 2023-01-28:
Closing due to inactivity

closed 2023-01-28 by technyon