technyon/nuki_hub #78: mqtt errors

isibizi, 2023-01-15:
Hello,

one more probelm :(

i loose the connection when i open the web ui, this i the log from the broker

`2023-01-15 01:57:47: New client connected from 10.5.0.43:53041 as nukihub (p2, c1, k15, u'mqtt_user').
2023-01-15 01:58:51: New connection from 10.5.0.43:52598 on port 1883.
2023-01-15 01:58:51: Client nukihub already connected, closing old connection.
2023-01-15 01:58:51: New client connected from 10.5.0.43:52598 as nukihub (p2, c1, k15, u'mqtt_user').
2023-01-15 01:59:01: New connection from 172.30.32.2:40612 on port 1883.
2023-01-15 01:59:01: Client <unknown> closed its connection.
2023-01-15 01:59:16: Saving in-memory database to /data//mosquitto.db.
2023-01-15 01:59:54: New connection from 10.5.0.43:52042 on port 1883.
2023-01-15 01:59:54: Client nukihub already connected, closing old connection.
2023-01-15 01:59:54: New client connected from 10.5.0.43:52042 as nukihub (p2, c1, k15, u'mqtt_user').
2023-01-15 02:00:58: New connection from 10.5.0.43:60153 on port 1883.
2023-01-15 02:00:58: Client nukihub already connected, closing old connection.
2023-01-15 02:00:58: New client connected from 10.5.0.43:60153 as nukihub (p2, c1, k15, u'mqtt_user').
2023-01-15 02:01:01: New connection from 172.30.32.2:45730 on port 1883.
2023-01-15 02:01:01: Client <unknown> closed its connection.
2023-01-15 02:02:01: New connection from 10.5.0.43:60727 on port 1883.
2023-01-15 02:02:01: Client nukihub already connected, closing old connection.
2023-01-15 02:02:01: New client connected from 10.5.0.43:60727 as nukihub (p2, c1, k15, u'mqtt_user').
2023-01-15 02:03:01: New connection from 172.30.32.2:60586 on port 1883.
2023-01-15 02:03:01: Client <unknown> closed its connection.
2023-01-15 02:03:05: New connection from 10.5.0.43:52333 on port 1883.
2023-01-15 02:03:05: Client nukihub already connected, closing old connection.
2023-01-15 02:03:05: New client connected from 10.5.0.43:52333 as nukihub (p2, c1, k15, u'mqtt_user').
2023-01-15 02:04:08: New connection from 10.5.0.43:54968 on port 1883.`

technyon, 2023-01-15:
Looks like a reboot loop. Do the logs stop when you close the browser?

technyon, 2023-01-20:
Please recheck with version 7.0

technyon, 2023-01-28:
Closing due to inactivity ... should be fixed in 8.0

closed 2023-01-28 by technyon