technyon/nuki_hub #79: [Web Interface Issue]

mdellagi77, 2023-01-16:
Thanks for this great project

I have followed the readme and 
1. flashed my ESP32
2. Setup the Wifi on it via the Window that Popped Up
3. Now I am trying to navigate to the esp32's web interface (Found the ip address in my router's client page)

But thats it. There is just nothing coming up. What am I doing wrong?

technyon, 2023-01-16:
Nothing, as long as you have the right IP address it should work. You could try adding http:// in front of the IP address, so the browser doesn't try a https connection.

mdellagi77, 2023-01-17:
Tried Firefox Developer, Chrome and Safari. With http, just ip. Nothing. Just a spinner of death

technyon, 2023-01-17:
It's really hard to tell why it doesn't work, especially if you were able to configure the Wifi in the first place. Can you ping the device? If possible, please connect to the serial line and post the logs (115200 Baud rate)

mdellagi77, 2023-01-17:
**Solved**. As my ESP-32 was 2.4G I connected it to the Guest Network which has rstictions on the LAN!

technyon, 2023-01-17:
ok that makes sense :)

closed 2023-01-17 by technyon