technyon/nuki_hub #80: home assistant problem

gatd4, 2023-01-17:
if I restart the home assistant computer home assistant does not see the nuki esp unless I restart the esp bridge?
I dont have to restart my other mqttt stuff.

KitHubek, 2023-01-17:
the same here. This problem is from i remeber I had version 6.6 and problem was too.

technyon, 2023-01-20:
Can you please recheck with version 7.0. My guess is that the discovery topics missed the retained flag, which should be fixed in 7.0.

KitHubek, 2023-01-20:
have 7.0 will write here after some days if problem exist still

gatd4, 2023-01-22:
still the same with v7
restart HA and nuki lock is unavailable

technyon, 2023-01-22:
I'm not really sure what would cause this. As a workaround you could automatically restart the ESP whenever home assistants starts up. There's an MQTT node to trigger the restart (maintenance/reset)

mundschenk-at, 2023-01-22:
What broker are you using? I haven't seen this with Mosquitto.

gatd4, 2023-01-22:
I am using mosquitto
it does not happen with any other mqtt entities

mundschenk-at, 2023-01-28:
Ah yes, I can confirm the issue when the host is restarted, even with EMQX. I'm assuming it is because the MQTT broker runs on the same machine (at least in my case). The discovery messages appear to have the retained flag (but perhaps not all?).

bcutter, 2023-02-07:
Any option to fix that on the Nuki Hub side? Some kind of automatic restart?
Initiating this from the HA side (e. g. on HA OS restarts - which is hard to detect as only an event for HA restarts (Core) exists - and that would be overkill) is a bit hacky and everyone has to implement this on his own...

technyon, 2023-02-08:
I'm neither sure exactly why this doesn't nor how I should detect if HA has restarted. If someone can enlighten me, I can do something about it. Do the autodiscovery topics need to be republished after HA has started? This should be handled by the retained already. And if so, how can a restart be detected?

mundschenk-at, 2023-02-08:
The problem occurs when the broker is restarted. It's possible this is an issue with persistence (broker configuration), but my other MQTT client with HA auto-discovery does not have this problem, so it must do something differently. I'll try to look into the differences between the two.

technyon, 2023-02-08:
What do you think about publishing the autodiscovery topics again after reconnect?

mundschenk-at, 2023-02-08:
Mhm, I experimented a bit, but actually the discovery topics seem to be fine. The ESP however entere into an automatic reboot loop, connecting and disconnecting from the broker again and again, toggling availability in HA. There might be a bug in the MQTT reconnect handling?

bcutter, 2023-02-11:
Currently I can't unfortunately.

closed 2023-04-22 by technyon

enkama, 2024-01-19:
Hey. Sorry for opening an old issue. But I still get this Issue to date. Pretty much it works when I start the nuki hub after I start Homeassistant. When I restart Homeassistant it shows as unavailable until I restart the Nuki hub. Then it works again.