technyon/nuki_hub #81: Much respect for this solution but......

isibizi, 2023-01-19:
Much resepct for this project but i have found for my usecase ab better and stabler solution than this project, maybe another can use this also.

First, i have installed this project as addon in Home Assistant.

The Name of the other project is **Virtual Nuki Bridge** what you can find here [RaspiNukiBridge](https://github.com/regevbr/RaspiNukiBridge) in combination with the HACS integration **hass_nuki_ng** is this in my eyes the best solution for HA. Nothing problems, fast response from the Nuki Smart Lock, everything works like charm and the best of all, you don't need an **esp32**

technyon, 2023-01-19:
I know there are other solutions, and they all have their advantages and disadvantages. Running on an ESP32 is quite an advance in my opinition, because:

- you don't need a raspberry pi :)
- An ESP is small and has low power consumption, you can place it pretty much anywhere where there's power. Makes it much easier to move it close to the lock

Since this isn't really an issue in the sense of a bug, I'll close this.

closed 2023-01-19 by technyon

isibizi, 2023-01-19:
This is why i wrote for for my usecase Jan-Ole :)

I have an Pi4 with HA and for me is this the best solution, because, i have allready an Pi and dont need anything else.