technyon/nuki_hub #82: Core Panic - (LoadProhibited)

tca22, 2023-01-19:
I'm using NukiHub 6.11 (ESP32 MiniKit V2.0)
there seems to be an unhandled Execption, causing a kernel-panic and subsequent reboot-loop:

[log attached]

not sure what triggers the core-panic, sometimes it's as soon as the web-ui is accessed, sometimes the reply by NukiOpener causes it;

another thing: the log shows, that the opener is paired, but it doesn't show on the WebUI;

Any ideas? 
[NukiHub-log.txt](https://github.com/technyon/nuki_hub/files/10456746/NukiHub-log.txt)

technyon, 2023-01-19:
Your ESP crashes, it's hard to tell why without debugging, so I can only give very general advice:

- check the power supply, eventually try a different USB cable
- Factory reset the ESP and flash again
- Try another ESP and see if you get the same error

closed 2023-01-19 by tca22

technyon, 2023-01-19:
That was actually a good hint. If the lock is disabled, the device crashes as soon as the browser tries to reload the web portal. Please check release 7.0 if it fixes your issue.

tca22, 2023-01-19:
yes, release 7.0 fixes above issue.
Thanks!!