technyon/nuki_hub #84: Extraneous topics (when certain accessories are not used)

mundschenk-at, 2023-01-19:
- `nuki/battery/keypadCritical`
- `nukiopener/lock/doorSensorState`

technyon, 2023-02-04:
The keypad node should be fixed in the latest release. Adding door sensor or not can be discussed. If not available, it should reflect this in the door sensor node, so it's not wrong to publish it.

iranl, 2024-02-11:
Door Sensor and Keypad entities will not appear in HA discovery if no door sensor or keypad is paired to the Nuki Opener/Lock respectively.

@mundschenk-at:
Is there still a request to remove/disable any possibly extraneous topics in MQTT?

Or is it ok to close this as won't fix?

mundschenk-at, 2024-02-11:
I will have a look at what is now published. Might take until next week though.

closed 2024-05-17 by iranl