technyon/nuki_hub #85: Not Pairing for me 🤷🏻‍♂️

mdellagi77, 2023-01-20:
1. I have flashed the esp32
2. I have opened the Web Interface and connected it to my Mosquitto MQTT server (Connected Yes)
3. I pressed 5 seconds on the Nuki 3 and the ring lit up 
4. I restarted the esp32

Exptected behavior: The esp32 will pair to the lock
Actual behavior: The status remains unpaired

Any ideas?

technyon, 2023-01-20:
The obvious thing to do is to make sure the ESP is close to the lock. Other than that have a look at this issue:

https://github.com/technyon/nuki_hub/issues/39

technyon, 2023-01-27:
Any update on this?

technyon, 2023-02-05:
Please recheck with the current release.

mdellagi77, 2023-02-06:
So. I received my new ESP32 from AlixExpress yesterday and it sync in like 5 sec. 
So you can close this issue.

Opening and closing via MQTT works no issues! Geat Project!! 🙇 

PS: Just a followup question. How does Autodiscovery work? I can send to topic "nuki/lock/action" "unlock" and "lock" now via the MQTT interface. But it seems you can skip setting up the mqtt lock on `configuration.yaml` with auto discovery? How do I discover to skip having to setup the lock and states manually?

mdellagi77, 2023-02-06:
Just maybe for making a list of supported esp32 with Nuki Pairing:

- XTVTX ESP32 ESP-WROOM-32 ✅ 
- AZDelivery ESP32 ESP-WROOM-32 (Failed for me 🤷)

technyon, 2023-02-06:
It probably worked for you now since I've downgraded the ble lib, gave me trouble with pairing too.

mdellagi77, 2023-02-06:
That would be an insane coincidence, but I have not crosschecked with the old esp32 again ;-)

closed 2023-02-24 by technyon

Dielee, 2023-03-02:
AZDelivery ESP32 ESP-WROOM-32 works for me!