technyon/nuki_hub #88: Too many messages from nuki hub

mahikeulbody, 2023-01-20:
Checking hass logs, I discovered a lot of messages from nuki hub with a very little interval of time. Please see an extract below.

2023-01-20 21:54:31.569 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-64'
2023-01-20 21:54:31.953 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/lock/rssi (qos=0): b'-70'
2023-01-20 21:54:32.530 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-63'
2023-01-20 21:54:35.100 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-64'
2023-01-20 21:54:37.105 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-63'
2023-01-20 21:54:38.098 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/lock/rssi (qos=0): b'-76'
2023-01-20 21:54:39.526 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-62'
2023-01-20 21:54:41.742 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-63'
2023-01-20 21:54:42.238 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/lock/rssi (qos=0): b'-70'
2023-01-20 21:54:43.843 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-62'
2023-01-20 21:54:45.302 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/lock/rssi (qos=0): b'-71'
2023-01-20 21:54:46.007 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-61'
2023-01-20 21:54:48.004 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-63'
2023-01-20 21:54:48.383 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/lock/rssi (qos=0): b'-70'
2023-01-20 21:54:50.013 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-64'
2023-01-20 21:54:52.037 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-63'
2023-01-20 21:54:54.050 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/maintenance/wifiRssi (qos=0): b'-62'
2023-01-20 21:54:54.398 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on nuki/lock/rssi (qos=0): b'-71'
2

What I can do to reduce this flow ?
That seems also to have an impact on the battery life of the smartlock.

mundschenk-at, 2023-01-21:
I also see very frequent updates for the RSSI topics (both WiFi snd Bluetooth, every 1 or 2 seconds!). The lock's battery level seems fine however (no rapid drain). I have not checked in the protocol level (HA would not update unchanged states).

technyon, 2023-01-21:
If you have battery drain, check the "Query interval" settings. The more frequently it is queried, the higher the battery drain. If this still a problem, let's put it into a seperate issue.

alexdelprete, 2023-01-21:
> What I can do to reduce this flow ?

You set the level to DEBUG for MQTT?

mahikeulbody, 2023-01-21:
If you mean the "Option: debug" of Mosquito addon, the answer is no.
Is there another place to check ?

mundschenk-at, 2023-01-21:
The [log level in HA](https://www.home-assistant.io/integrations/logger/) - unless this is a general development installation, it should not be set to `debug`.

mahikeulbody, 2023-01-21:
I have just "default_config:" into my configuration.yaml file, I have no logger defined.

mundschenk-at, 2023-01-21:
Maybe it is in some include file or package, it is obviously set to `debug` at least for the `mqtt` component.

mahikeulbody, 2023-01-21:
I unchecked Enable MQTT logging" into nuki hub configuration but I have the same messages.
nuki hub is constantly refreshing WiFi and Bluetooth signal strength so there will be too much messages, anyway.

mundschenk-at, 2023-01-21:
`nuki_hub`'s MQTT logging has got nothing to do with the fact that your Home Assistant installation has enabled debug logging for its `mqtt` component. The logged lines in HA's log are a result of your HA configuration and nothing else. The only thing `nuki_hub` does is trigger the issue frequently because of excessive RSSI topic updates.

mundschenk-at, 2023-01-21:
> I have no logger defined into my configuration.yaml file,

You might also have set it via the GUI (newer versions of HA allow you to that from the integrations menu).

mahikeulbody, 2023-01-21:
You are right : I just unchecked from the integration menu ; debug was set (I don't remember to have set it but who know...).

So this issue is resolved but I will open a new related to your #1 (Nuki Hub 7.0 contains a bug that triggers RSSI updates too often)

closed 2023-01-21 by mahikeulbody