technyon/nuki_hub #89: Nuki Hub triggers RSSI updates too often.

mahikeulbody, 2023-01-21:
(checked with v7)

nuki hub is constantly retrieving wifi and bluetooth signal strength.

An option to set a query interval for these values (as for battery) would be welcome.

technyon, 2023-01-21:
Is this really a problem? It's updating ~3 seconds, but you have an up-to-date value like this. There's no battery drain on the lock due to this, as the ESP forwards the strength of the signals it receives anyways.

mahikeulbody, 2023-01-21:
Indeed, if there is not battery drain that is probably not a problem except may be when you set DEBUG Mqtt, no matter the reason. : the log file is flooded of to many useless messages.

mundschenk-at, 2023-01-21:
From what I remember, it was more like 10-20 seconds before - is this just controlled by the locks energy-saving setting, or did something change with Nuki Hub since 6.x.x? Anyway, I can see two reasons why this might be problematic:

1. Every value change is stored in HA's database, potentially necessitating a write operation (you can mitigate this by changing `recorder` settings, but the tradeoff is dataloss in case of a power cut, and most people probably won't change the defaults).
2. There's quite a bit of network traffic for something that is by definition mostly noise (fluctuations in RSSI). 

Because of 2., I think it would be best if the last `n` values were averaged to smooth out noise, or maybe everything within in a certain time period (to get to a more regular "heartbeat")? This probably could be done on the autodiscovery/HA side, but that would not alleviate the network traffic.

technyon, 2023-01-22:
I've made it configurable in Release 7.1. It defaults to disabled, so if you still want updates, you have to configure it.

closed 2023-01-22 by technyon

alexdelprete, 2023-01-22:
> I've made it configurable in Release 7.1. It defaults to disabled, so if you still want updates, you have to configure it.

Thanks a lot for this.

technyon, 2023-01-23:
The check didn't work properly for the opener, it's fixed in 7.2