technyon/nuki_hub #90: Binary data in MQTT messages with 7.0

mundschenk-at, 2023-01-22:
With Nuki Hub 7.0, messages that are supposed to have a text payload sometimes contain additional (?) binary "data". Here are logs from my HA instance from today:

```
Can't decode payload b'1\xf3\x01' on nukiopener/lock/rssi with encoding utf-8 (for <Job HassJobType.Callback <function MqttSensor._prepare_subscribe_topics.<locals>.message_received at 0x7f6119a830>>)
Can't decode payload b'1\xf4\x01' on nuki/lock/rssi with encoding utf-8 (for <Job HassJobType.Callback <function MqttSensor._prepare_subscribe_topics.<locals>.message_received at 0x7f61198c10>>)
Can't decode payload b'1\xd1\x01' on nuki/lock/rssi with encoding utf-8 (for <Job HassJobType.Callback <function MqttSensor._prepare_subscribe_topics.<locals>.message_received at 0x7f61198c10>>)
Can't decode payload b'1\xcd\x01' on nukiopener/lock/rssi with encoding utf-8 (for <Job HassJobType.Callback <function MqttSensor._prepare_subscribe_topics.<locals>.message_received at 0x7f6119a830>>)
```

I've also added a logging automation to write `nuki/maintenance/log` messages to a file. Among the expected log output, it contained some weird lines looking like buffer problems again:

```
2023-01-22 04:10:43.636637+01:00: MQTT connected
2023-01-22 04:28:41.145377+01:00: MQTT connected
2023-01-22 04:28:56.546967+01:00: MQTT connected
2023-01-22 04:29:05.253025+01:00: 1 nuki/lock/r
2023-01-22 04:29:05.460348+01:00: MQTT connected
2023-01-22 05:06:57.424320+01:00: Reading config. Result: success
2023-01-22 05:06:57.631183+01:00: Reading advanced config. Result: success
```

mundschenk-at, 2023-01-24:
Yeah there's a whole dictionary of keywords for auto discovery.

Mincka, 2023-01-25:
Got this buggy one this morning on 7.2, 16 minutes after a restart:
[image]

technyon, 2023-01-25:
That's a valid state. From the API documentation:

0x06 unlocked (lock ‘n’ go active)

Mincka, 2023-01-25:
Lol, sorry about that. Good catch! I can confirm I triggered Lock'n'Go at that time. So no problems here.

mundschenk-at, 2023-01-28:
To document progress: The issue appears to be fixed with the [`espmqttlib` branch](https://github.com/technyon/nuki_hub/tree/espmqttlib).

technyon, 2023-01-28:
Should be fixed in 8.0

closed 2023-01-28 by technyon