technyon/nuki_hub #91: MQTT not connecting : password length

NOYB4Europe, 2023-01-22:
I just wanted to let you know, that I was able to resolve my "MQTT not connecting" issue. 
It appears that a MQTT password with 65 digits is too long. 27 digits did work for me! 

Thanks for your work!

technyon, 2023-01-22:
Well it's a microcontroller and memory is best allocated statically, so I had to limit the length somewhere, 30 characters is max.  It's not a bug, it's feature.

Isn't 65 characters a bit over the top anyway? If you want some real security, you should use certificates to that all traffic is encrypted.

closed 2023-01-22 by technyon

NOYB4Europe, 2023-01-22:
Hi technyon: 65 is over the top, must have been my cat walking over the keyboard. nevertheless it took me some hours until i had figured out what the issue was! :-)

technyon, 2023-01-22:
Yes it' not too obvious. The password field only accepts 30 characters, but you can't see it because they are all masked out. I'll add a not in the interface.