technyon/nuki_hub #92: NukiKeypad not seen by Nuki Hub 7.1

tca22, 2023-01-23:
I am using NukiOpener, NukiKeypad and Nuki Hub 7.1 .

There are MQTT messages for NukiOpener (nukihub:nukiopener) but nothing for NukiKeypad. There is no "Enabled keypad control via MQTT" in the WebUI.

Did I miss something? Should it be possible to manage NukiKeypad through Nuki Hub 7.1?

technyon, 2023-01-23:
The keypad is unfortunately only implemented for the lock at this time.

tca22, 2023-01-23:
ok - kindly request this feature for a future update...

closed 2023-01-23 by tca22

knorrre, 2023-01-26:
Same problem here: There is no "Enabled keypad control via MQTT" in the WebUI.
https://i.imgur.com/54suCV5.png

cpvbruno, 2023-01-31:
Hey, sorry, it's not clear to me, in the readme file says the keypad can be managed using the hub but I also don't see "Enabled keypad control via MQTT" in the web UI, is the readme file not updated?

technyon, 2023-02-07:
Check the latest release, keypad for opener is added.

tca22, 2023-02-10:
Thanks technyon!!

Not sure if I missed something, but is there any checkbox to select?
I'm  on V8.7 (with Nuki-Opener & Nuki-Keypad) but there is no MQTT message from the keypad coming through, nor can I see something on FHEM [MQTT2_DEVICE].
Any thoughts?

technyon, 2023-02-10:
It's a bug unfortunately. You only see they option if the keypad is connected to the lock. It'll be fixed in the next release.

In the meantime you should be able by passing the correct parameter in the URL like this:

http://<NUKI_HUB_IP>/method=get?KPENA=1

tca22, 2023-02-11:
Works! Thank you!!!

tca22, 2023-02-11:
kindly ask for clarification:

Are the MQTT-topics "battery voltage" and "battery critical" used for the keypad as well?
Should they appear as "nukihub:nukiopener/keypad/battery/..."?
Not sure if NUKI-API provides this information in first place, so it might not be possible to forward at all...

technyon, 2023-02-11:
You'll find it under battery/keypadCritical. For the keypad there's no level indication, only critical or not.