technyon/nuki_hub #94: ESP32-C3 Support?

Diddlik, 2023-01-25:
During the installation of ESP32-C3 I got an error message that it is not compatible with C3. Will it be compatible later or is it technically not possible?

Diddlik, 2023-01-25:
my idea was to design out of box device like nuki brige. You put it into the socket and use it with nuki hub. My choice were esp32 c3, because it is the smallest esp32 module.
I will try to redesign it with esp32-s.
Thank you for explanation.

technyon, 2023-01-25:
While what I said still holds true, we can give it a try. This is a binary compiled for the C3. I have idea wether or not it works, since I don't have C3 to test it with. 

What about using a D1 mini form factor:

https://www.amazon.de/AZDelivery-Bluetooth-Internet-Development-Compatible/dp/B08BTLYSTM

It's still quite small, I've used in a project where I had limited space.

[nuki_hub-c3.zip](https://github.com/technyon/nuki_hub/files/10501757/nuki_hub-c3.zip)

Mincka, 2023-01-25:
If you want something small, compatible and with a case, have a look at these:
https://shop.m5stack.com/products/atom-lite-esp32-development-kit?variant=32259605200986

Almost the same size than the smallest C3 (24*24*10mm) with more features.

Diddlik, 2023-01-25:
> If you want something small, compatible and with a case, have a look at these: https://shop.m5stack.com/products/atom-lite-esp32-development-kit?variant=32259605200986
> 
> Almost the same size than the smallest C3 (24*24*10mm) with more features.
> 
> With with to plug it on a phone charger if you don't want to use a short USB cable: https://www.amazon.fr/3-1-m%C3%A2le-2-0-adaptateur-Chromebook-tablette-appareils/dp/B075B1T4Y9
> 
> The whole footprint is smaller than the Nuki Hub.

I have thought about m5stack, unfortunately I have not found a good solution how it can combine well with the socket.

Diddlik, 2023-01-25:
> While what I said still holds true, we can give it a try. This is a binary compiled for the C3. I have idea wether or not it works, since I don't have C3 to test it with.
>
> What about using a D1 mini [...]

Thanks, I will try your build. 
i am currently waiting for the power supplies from aliexspress. Then I will see how I can pack everything well and compactly together

Mincka, 2023-02-02:
Nice job! Will you share the parts and sources for both PCB and case?

Diddlik, 2023-02-02:
> Nice job! Will you share the parts and sources for both PCB and case?

Yes, after all testing are done and I can confirm that my approach is ok, I will share all information.

yehoolte, 2023-02-14:
> 

Hi Technyon!

Thank you for this great project!
I've tested the c3 bin file, looks like flash and download completed with no errors
Now, I don't see wifi network, am I missing something?
[image]

[image]

[image]

Thanks!

Diddlik, 2023-02-17:
> > 
> 
> Hi Technyon!
> 
> Thank you for this great project! I've tested the c3 bin file, looks like flash and download completed with no errors Now, I don't see wifi network, am I missing something? [image]
> 
> [image]
> 
> [image]
> 
> Thanks!

I have a boot loop with c3

technyon, 2023-02-17:
Hi,

@yehoolte The C3 is not supported. The binary was just for testing if it could work. Since the experiment failed, please use a regular ESP32.

yehoolte, 2023-02-19:
@technyon 
Thanks anyway! I will use the ESP32

Diddlik, 2023-02-23:
[image]

Hi Guys, the pcb is here. I will perform some tests. Enclosure is designed as well 😁

Diddlik, 2023-03-14:
I think this topic can be closed. I have uploaded an esp32 pcb and small case for it to github 
https://github.com/Diddlik/Esp32-Mini-PCB

technyon, 2023-03-18:
Amazing job! I'll add to the readme.

closed 2023-03-18 by technyon

werty1st, 2023-04-01:
Hi. I tried the last few hours to build this on MacOS, without luck. (the VM was to big to download from mega.) next time I try it in my Ubuntu VM. 
I have a few M5 stamp c3 and I was able to compile wled for them. so I hope to get this software running too.

technyon, 2023-04-02:
I think the toolchain has some problems on MacOS. Why do you need it running on a C3 ?

werty1st, 2023-04-02:
Because I have 6 of the m5StackC3.

I tried to follow the instructions from here: https://github.com/technyon/coffeetimer

then downloaded the last 1.x Arduino Release.

Installed:
```bash
sudo apt-get install git wget make libncurses-dev flex bison gperf python3 python3-serial

sudo apt install build-essential cmake clang cmake-curses-gui
```
and the ESP toolchain from here: https://docs.espressif.com/projects/esp-idf/en/release-v3.0/get-started/linux-setup.html

In Ardurino I installed the ESP32 Software 2.0.7

ran:
```bash
export PATH="$PATH:$HOME/Downloads/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0/xtensa-esp32-elf/bin/"
export ARDUINO_SDK_PATH="/home/ubuntu/Downloads/arduino-1.8.19/"
#export CXX="/usr/bin/clang++" #because of the following error

cmake \
-D CMAKE_TOOLCHAIN_FILE=/home/ubuntu/Downloads/Arduino-CMake-Toolchain/Arduino-toolchain.cmake \
-DARDUINO_INSTALL_PATH="/home/ubuntu/Downloads/arduino-1.8.19/" .
```

and also:
```bash
./nuki_hub$ ccmake .
```

> The CXX compiler identification is unknown
>  CMake Error at CMakeLists.txt:5 (project):
>    The CMAKE_CXX_COMPILER:
> 
>      {tools.{build.tarch}-esp32-elf-gcc.path}/bin/xtensa-esp32-elf-g++
> 
>    is not a full path and was not found in the PATH.
> 
>    Tell CMake where to find the compiler by setting either the environment
>    variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
>    to the compiler, or to the compiler name if it is in the PATH.

I don't know how to fix that problem.

technyon, 2023-04-04:
You should use my version of the toolchain:

https://github.com/technyon/Arduino-CMake-Toolchain

I've updated it in the coffetimer readme.

werty1st, 2023-04-04:
thank you.
but i have finally downloaded the vm and will try it first.

alexdelprete, 2024-06-30:
> I think this topic can be closed. I have uploaded an esp32 pcb and small case for it to github https://github.com/Diddlik/Esp32-Mini-PCB

Amazing work. Do you sell prebuilt devices somewhere? I'm really bad with hw unfortunately but would love to have this. Please let me know, thanks.